ev3dev/ev3dev #1446: python pip doesn't work

Znunu, 2020-10-28:
**Describe the bug**
I'd like to use pip. Tried to install it, but then it doesn't work,

**To Reproduce**
Steps to reproduce the behavior:
1. `sudo apt-get install python3-pip`
2. `pip3 --version`

**Expected**
Version information is printed

**What happens**
Nothing is ever printed and pip simply run/hangs until `CTRL+C`

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535DB143
BOARD_INFO_TYPE=main
```

dlech, 2020-10-28:
Python is very slow on the EV3. You probably didn't wait long enough before CTRL+C. Also, we highly recommend installing Python packages with `apt` rather than `pip`. E.g. `sudo apt update && sudo apt install python3-numpy`.

Znunu, 2020-10-30:
Wow, you were completely right. I think it took like a minute. I'm wondering why so long. Even on my PC it's like a second or two.

I tried with apt, but to no avail since it didn't have the package that I needed (discord.py). It _did_ have the largest dependency of discord.py, aiohttp. Unfortunately a version too old for discord.py. So I gave ev3 a fresh set of batteries, entered `pip3 install discord.py` and went to sleep. When I woke up, it was off, discord.py wasn't installed and the batteries were used up. My final solution was to build the library on another machine and copy it over, which fortunately seems to be working :)

btw thanks so much for helping out with and responding to everyone who has issues. A lot of repositories have maintainers who don't seem to care ;-;

closed 2020-11-09 by Znunu