ev3dev/ev3dev #1448: Unable to set up Network Sharing & SSH because it keeps saying the cable is disconnected

Arjdroid, 2020-11-05:
> So Far, My Objective Has Just Been To Be Able To Connect To My EV3 Brick Via SSH And Be Able To Connect To The Internet From My EV3 Brick.

**What was I trying to do?**

I was trying to set up a wired network connection and share my Computer's Internet Connection with the EV3 brick.

**What did I expect to happen?**

My operating systems should have detected that the EV3 brick is plugged in and allow me to configure the appropriate network sharing settings.

I have tried it on a MacOS Catalina System, Windows 10 20H2 System and an Ubuntu 20.10 system but none of them work. I think the instructions on your documentation are severely outdated and they may be contributing to my issues...

On MacOS 10.15.17 (Catalina) -

[image]

On Windows 10 20H2 (October 2020 Update) -
The device IS being detected by Windows in _Control Panel > Show Printers & Devices_

[image]

But even after I set up the Network Sharing on my main adapter, in _Control Panel > Network Adapters_, it is saying the Cable is Unplugged

[image]

What SEEMS to be working is when I go into my EV3 Brick's settings and then go to Wifi & Networks While Connected To My EV3 Brick And In The EV3 Brick Settings I go to Tethering and Select The Gadget Option. Then on the top left corner of the EV3 Screen, an IP Address pops up. On my PC, when I ssh into robot@ev3dev, enter the password maker, a Bash Prompt appears but the problem is that the network is unreachable. Even though I set up the Network sharing option on my adapter with the EV3, it is unable to reach outside of the tether. The problem is that when there is no tether, neither of the devices see each other even though the PC can see that the EV3 is connected. The EV3 says that there are no connections when according to the Instructions, it should be saying that there is a Wired Connection...

dlech, 2020-11-05:
Which version of ev3dev are you using? Can you give the exact steps needed to reproduce the problem?

R2D2Hud, 2020-11-05:
You could have a bad usb cable.

Arjdroid, 2020-11-06:
Ok guys, thank you for all the help, I reflashed the O.S and that seemed to fix the problem, I was using the latest O.S from the releases page. Only thing I changed was that I used etcher on Mac instead of Windows to flash the O.S, that seems to have fixed the problem. Again, thank you for all the help!

closed 2020-11-06 by Arjdroid