ev3dev/ev3dev #1450: How to install a new laibrary on EV3

kentaro-wada, 2020-11-15:
I'm using a micro-python(ev3dev).

I want to install a new library(
pip install pyzbar
pip install pillow
) on EV3.

But I don't know how to install it.

If you know how, please tell me.
Thanks in advance.

kentaro-wada, 2020-11-15:
This is my result on terminal in VSC.

robot@ev3dev:~$ sudo apt install pyzbar
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package pyzbar
robot@ev3dev:~$ 

What is the cause of "E: Unable to locate package pyzbar"?

dlech, 2020-11-15:
Python is not the same as MicroPython, so Python packages will only work with regular Python. Python packages in Debian usually start with `python-` for Python 2 and `python3-` for Python 3. So `sudo apt update && sudo apt install python-zbar` will install the package. However there isn't a Python 3 version, so you won't be able to use it in the same program with the `ev3dev2` library since that is Python 3 only.