ev3dev/ev3dev #1455: connect two EV3's (both ev3dev) via bluetooth and transfer data

JanF-04, 2020-12-23:
How to transfer / send and receive data using bluetooth?

I am trying to connect two EV3 robots via bluetooth to transfer information. To establisch the connection is very easy using the GUI on the robots. But how do you send and receive the data? Additionally I researched, the BT Address should be the same as the Board Serial Number, but it isn't. Running the command `hciconfig` shows the address is ´F4:84:4C:CA:6F:3A´.
**Output:**
```
hci0:   Type: Primary  Bus: UART
        BD Address: F4:84:4C:CA:6F:3A  ACL MTU: 1021:4  SCO MTU: 180:4
        UP RUNNING PSCAN INQUIRY
        RX bytes:339117 acl:17 sco:0 events:2442 errors:0
        TX bytes:12811 acl:16 sco:0 commands:615 errors:0
```

 - Connection Type: bluetooth
 - Device: both robots are EV3 robots running ev3dev
 
```
Image file:            ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:            0.10.3
BogoMIPS:          148.88
Bluetooth:           4.0
Board:                 board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016536451A9
BOARD_INFO_TYPE=main
```

dlech, 2020-12-23:
Have a look at https://pybricks.github.io/ev3-micropython/messaging.html. It does all of the hard work for you, including letting you reference devices by name rather than Bluetooth address.

JanF-04, 2020-12-23:
Thanks for helping, but I want to use Rust (https://github.com/pixix4/ev3dev-lang-rust) for programming the robot. It is a school project and we should not change the programming language. Does anybody know if there is a crate for Rust too?

Eisverygoodletter, 2022-09-23:
Is there new information on this? I'm interested to know.