ev3dev/ev3dev #1456: Bluetooth communication with computer

yusufziy4, 2020-12-24:
How can I comminucate between my computer using python and ev3.

Also I can use 

OS: Windows 10,
Connection Type: Bluetooth,
IE: VSCode
Device: EV3

System Info:
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653623786
BOARD_INFO_TYPE=main

I dont why there is no bluetooth but I am pretty sure I did use it once.

dlech, 2020-12-24:
Bluetooth is probably turned off on the EV3.

Fabs37, 2021-04-11:
When there's just "not available" on the screen when you open the Bluetooth settings on the Brick, then reboot the bot.
General tutorial on Bluetooth connection with ev3dev: https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/