ev3dev/ev3dev #1457: how to update linux wifi driver for edimax adapter on ev3dev

aerosquirrel, 2020-12-28:
**How do you update the Edimax WiFi adapter drivers on Ev3Dev?**

**Background Information**
I am trying to develop a remote robotics/telerobotics capability for the EV3 so students can log in, program, and operate robots that are located in venues around the world (for education during Covid and after).  I'm taking the approach of networking the robots over wifi.  One of the challenges is that the officially approved wifi adapters for the EV3 are discontinued (no longer manufactured).

The approved (discontinued) Edimax EW-7811Un wifi adapter, which has the smallest profile, has been replaced by Edimax with EW-7811Un V2, which is available.  It has a new chipset and new drivers, which exist for Linux.  To make this new adapter work, I believe I need to update the wifi drivers on the EV3.  

How do I update the drivers (and can I update the drivers) in Ev3Dev and/or native Ev3 firmware?

Here is the updated driver for the Edimax E@-7811Un V2.   
https://www.edimax.com/edimax/download/download/data/edimax/global/download/wireless_adapters_n150/ew-7811un_v2

**Desktop (please complete the following information):**
 - OS and version: Windows 10 
 - Connection Type: WiFi
 - Software: 

**Robot:**
 - Device: EV3
 - Versions: Native FW:  V1.09E / EV3Dev v200?

xdsl, 2021-01-18:
https://github.com/ev3dev/ev3dev/issues/1442#issuecomment-757471893

aerosquirrel, 2021-01-26:
@xdsl Thanks!  I'm working through the details now, trying to implement it.

Going back to the very basics, what's the most efficient way to transfer these files into ev3dev from windows?  NFS?  Setting up FTP?  A command through SSH?  Please advise.  (My raspi was kind enough to support FTP, so I'm a little befuddled here).

I think given the complexity I'm going to add the drivers and make a custom EV3dev image for my users so they don't have to figure out the upgraded driver process.  Is there a way to campaign for an updated EV3dev OS with new drivers included?

Thanks!

xdsl, 2021-01-27:
> Going back to the very basics, what's the most efficient way to transfer these files into ev3dev from windows? NFS? Setting up FTP? A command through SSH? Please advise. (My raspi was kind enough to support FTP, so I'm a little befuddled here).

You can use putty and pscp: https://www.poftut.com/how-to-use-pscp-command-on-windows/

> I think given the complexity I'm going to add the drivers and make a custom EV3dev image for my users so they don't have to figure out the upgraded driver process. Is there a way to campaign for an updated EV3dev OS with new drivers included?

https://github.com/ev3dev/ev3dev/issues/1458#issuecomment-753652773

ihormudryy, 2021-02-01:
Hi, wouldn't it be easier to compile the kernel with a new driver for Edimax EW-7811Un V2? There is one in menuconfig, but it's disabled by default and I believe its source code has to be updated in the lego-linux-drivers repo.

xdsl, 2021-02-02:
it's very right way for lego-linux-driver developers ... but for now i suggest a working solution - https://github.com/ev3dev/ev3dev/issues/1442#issuecomment-757471893

ihormudryy, 2021-02-02:
Yes, but this is very frustrating when you take the latest and greatest statch image from ev3dev.org, buy recommended wi-fi dongle and it simply doesn't work out-of-the-box. Then you spend time struggling to find out what chip is it and where is the driver, then compile it using arm toolchain and then you find a binary on Github :) I'm ready to contribute to making changes for a new image with embedded drivers so next users won't suffer as I did.

xdsl, 2021-02-10:
I generate ev3dev-stretch-ev3 image from ev3dev/ev3dev-stretch-ev3-generic for my RubiRobot project (http://rubirobot.ru/). Disabled ipv6 and many unused services, added Free Pascal, GCC, Midnight Commander, ttf-dejavu, RubiRobotLib 0.2.6 and new kernel with support  Edimax EW-7811Un V2. If you want just use it: http://rubirobot.ru/files/img/ev3dev-S-rubirobot.img.xz (456Mb), md5: http://rubirobot.ru/files/img/ev3dev-S-rubirobot.img.xz.md5.txt

cho934, 2021-03-03:
Hi,

How to generate/build easily a custom ev3-stretch kernel  with a wifi module  (like the rtl8812au which is in fact inside the official release of ev3dev-kernel (stretch) ? Could you describe how to do it ?

prasun-raha, 2021-03-30:
@xdsl , I am trying to use the nl80211 driver for the edimax EW-7811Un V2. I am using your build but I do not see the nl support for the driver. it works with the iwlist but not with iw. Whats the best way to add this, I will need this to make the repo below work with your image as the starting point
 https://github.com/bmegli/ev3dev-mapping-modules
Thanks!

xdsl, 2021-03-31:
@prasun-raha, I'm not a wifi driver developer)
I am just using docker to compile the ev3dev-stretch kernel with the latest rtl8188eu driver from https://github.com/lwfinger/rtl8188eu and use it in my RubiRobot project. If you like, I can export this docker image for you.

xdsl, 2021-03-31:
@prasun-raha, does this problem appear on the original ev3dev-stretch kernel?

prasun-raha, 2021-03-31:
> @prasun-raha, does this problem appear on the original ev3dev-stretch kernel?

I could not get the rtl8188eu driver working on the original ev3dev-stretch, hence I was using the image you provided directly - that seems to work fine for me

xdsl, 2021-03-31:
may be help: 
https://www.spinics.net/lists/hostap/msg07611.html
https://github.com/lwfinger/rtl8188eu/tree/v5.2.2.4
In my image I used master branch rtl8188eu, v4.1.4: https://github.com/lwfinger/rtl8188eu

DrMeers, 2023-01-15:
Flashing the http://rubirobot.ru/files/img/ev3dev-S-rubirobot.img.xz image from @xdsl on the SD card worked perfectly for me just now with a v2 adapter that was failing on stock ev3dev (and standard firmware), thanks @xdsl !

BigKitten, 2023-08-29:
I am not able to open the website to download the image?

audricchabert, 2024-05-25:
same as @DrMeers , the image from @xdsl  worked perfectly, thanks!

Growflavor, 2024-08-19:
> So, there is no dongles left for the original Lego EV3 firmware?

For me on our EV3 bricks, this specific wifi adapter with BOTH EV3G (= OEM LEGO firmware 1.10E) & EV3dev...
I have purchased 2 of them over the past few weeks in the US:

https://www.amazon.com/dp/B008O2AL0K

[image]

dlech, 2024-08-19:
> For me on our EV3 bricks, this specific wifi adapter works with BOTH EV3G (= OEM LEGO firmware 1.10E) & EV3dev...
> I have purchased 2 of them over the past few weeks in the US.

What is the output of `lsusb` for this dongle?

Growflavor, 2024-08-19:
> > For me on our EV3 bricks, this specific wifi adapter works with BOTH EV3G (= OEM LEGO firmware 1.10E) & EV3dev...
> > I have purchased 2 of them over the past few weeks in the US.
> 
> What is the output of `lsusb` for this dongle?

robot@ev3dev:$ lsusb
Bus 001 Device 002: **ID 0bda:8176** Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
robot@ev3dev:$ 

Windows shows it also as a rev_0200:
[image]

So, the two netis wifi adapters I received have:  ID 0bda:8176 rev_0200

dlech, 2024-08-19:
Thanks for the info. This looks like a re-branded version of a known working adapter. I've updated the [wiki](https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles) with this info.

Growflavor, 2024-08-20:
> Thanks for the info. This looks like a re-branded version of a known working adapter. I've updated the [wiki](https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles) with this info.

RE: the following EV3dev tutorial www page
https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/
https://github.com/ev3dev/ev3dev.github.io/blob/master/docs/tutorials/connecting-to-the-internet-via-bluetooth.md

One other EV3 internet related & useful wiki/www site update is that Windows 10 & 11 actually do now support Bluetooth connection access to the PC internet via a "Mobile Hotspot".

I shared some Win10 & Win11 screen snips showing the relevant windows settings/control panels at:
https://github.com/orgs/ev3dev/discussions/1646#discussioncomment-10376087

& as an example, here is an EV3 git pull (from https://github.com/K2x141b/pathfind3r ) done via a Bluetooth connection with a "Mobile Hotspot PC ethernet" (Windows 11 in this case): 

[image]