ev3dev/ev3dev #1458: How to fully disable ipv6 on ev3dev?

xdsl, 2021-01-02:
How to fully disable ipv6 on ev3dev?

Platform: Lego Mindstorms EV3

```
>uname -a
Linux ev3dev 4.14.117-ev3dev-2.3.5-ev3 #1 PREEMPT Sat Mar 7 12:54:39 CST 2020 armv5tejl GNU/Linux

```

dlech, 2021-01-02:
I would guess the same way you disable it on any other Linux machine.

xdsl, 2021-01-02:
found: in /`boot/flash/uEnv.txt` set `bootargs=ipv6.disable=1`
but sit0 interface remains, how to disable it?
```
>ip -a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
...
```

dlech, 2021-01-02:
I suppose you could compile your own kernel without IPv6. https://github.com/ev3dev/ev3dev-buildscripts/tree/ev3dev-stretch

xdsl, 2021-01-03:
My project is based on mainline kernel. 
I don't want to give up ev3dev update with apt-get dist-upgrade.
Оr 4.14.117-ev3dev-2.3.5-ev3 is the latest kernel on ev3dev-stretch?

dlech, 2021-01-03:
What is your project? 

4.14.117-ev3dev-2.3.5-ev3 is the latest kernel on ev3dev-stretch

xdsl, 2021-01-04:
> What is your project?
http://rubirobot.ru/
Free Pascal and Lazarus on Linux for programming Lego Mindstorms EV3.

xdsl, 2021-01-05:
> I suppose you could compile your own kernel without IPv6. https://github.com/ev3dev/ev3dev-buildscripts/tree/ev3dev-stretch

Can I use Ubuntu in docker for compile kernel?

dlech, 2021-01-05:
Probably. I have never tried it. Windows Subsystem for Linux should work too if you are on Windows. Just be sure that the kernel source code is not checked out on an NTFS file system. The lack of case sensitivity will cause failure.

xdsl, 2021-01-10:
> Probably. I have never tried it. Windows Subsystem for Linux should work too if you are on Windows. Just be sure that the kernel source code is not checked out on an NTFS file system. The lack of case sensitivity will cause failure.

I do not use Windows. 
I have tested cross-compiling ev3dev kernel and some modules on Ubuntu in docker. It works.
Unfortunately Brickd doesn't work without ipv6. Message:
```
Jan 09 19:04:22 ev3dev brickd[413]: brickd.vala:58: Board info not found
Jan 09 19:04:22 ev3dev brickd[413]: Failed to start: Unable to create socket: Address family not supported by protocol

```

xdsl, 2021-01-27:
https://github.com/ev3dev/ev3dev/issues/1442#issuecomment-764498324

closed 2021-01-27 by xdsl