ev3dev/ev3dev #1460: How to process values from Temperature Sesnor

jabrena, 2021-01-06:
Hi @dlech,

I am trying to implement the support for NXT Temperature Sensor and I would like to understand the values from ev3dev.
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensor_data.html#lego-nxt-temperature-sensor

Currently, I receive the following output from the sensor:

```
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat mode
NXT-TEMP-C
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat modes
NXT-TEMP-C NXT-TEMP-F
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat units 
C
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat value0
165
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat decimals 
1
```

What is the measing of 165? 
1.65 celsius degrees?

The sensor has a Temperature range: (-550 to 1280)

Do I need to do an intermediate calculus?

Many thanks in advance

Juan Antonio

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           226.09
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165335B6C5
BOARD_INFO_TYPE=main
```

dlech, 2021-01-06:
Since `decimals` is 1, it means add one decimal place (divide by 10). So `value0` of 165 with `mode`NXT-TEMP-C` means 16.5 °C.

jabrena, 2021-01-06:
Many thanks @dlech 

I will process in that way

Happy new year

Juan Antonio

closed 2021-01-06 by jabrena