ev3dev/ev3dev #1461: Sudo apt update

Aromik, 2021-01-06:
Hello!
Today I have installed the latest version of ev3dev. When entering in console
`sudo apt update`
I see this error:
`Err:1 http://archive.ev3dev.org/debian stretch InRelease
  Could not resolve 'archive.ev3dev.org'
Err:2 http://httpredir.debian.org/debian stretch InRelease
  Could not resolve 'httpredir.debian.org'
Err:3 http://security.debian.org stretch/updates InRelease
  Could not resolve 'security.debian.org'
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
W: Failed to fetch http://httpredir.debian.org/debian/dists/stretch/InRelease  Could not resolve 'httpredir.debian.org'
W: Failed to fetch http://security.debian.org/dists/stretch/updates/InRelease  Could not resolve 'security.debian.org'
W: Failed to fetch http://archive.ev3dev.org/debian/dists/stretch/InRelease  Could not resolve 'archive.ev3dev.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`
What can I do to fix it?
Thank you!

dlech, 2021-01-06:
Make sure the EV3 is connected to the Internet. Instructions: https://www.ev3dev.org/docs/networking/

Aromik, 2021-01-06:
It is connected because I work with ssh. I setuped a Wi-Fi via Bluetooth connection

dlech, 2021-01-06:
A network connection is not the same as an Internet connection. So it is possible to be connected with SSH but still not be connected to the Internet. If you are using Windows for Internet Connection Sharing, it is not possible to connected the EV3 to the Internet using Bluetooth.

closed 2021-01-06 by Aromik