ev3dev/ev3dev #1462: how to remove ev3dev and reinstall the official LEGO firmware?

Amplifon1, 2021-01-13:
**Question**

How do you ...?

**Background Information**

I am trying to ... because I want to build a robot that ...

**Screenshots**
If applicable, add screenshots to help explain your question.

**Desktop (please complete the following information):**
 - OS and version: [e.g. Window 10 1809, macOS 10.14]
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - Software: [e.g. Visual Studio Code, PuTTY]

**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

dlech, 2021-01-13:
Turn off the EV3 and remove the SD card. The original firmware was never removed.

Amplifon1, 2021-01-13:
Thanks, that was my mistake.
The issue is solved

closed 2021-01-13 by dlech