ev3dev/ev3dev #1463: Run two motors simultaneously in very small steps.

MrNoZzz, 2021-01-20:
Hi. 
I started on my project to build a 3D printer in LEGO useing EV3 Dev and programing i Python. The Mechanical part of the printer is going quite well. But as a mechanical ingeer, programing is not my strong suit. I am in the process of optimizing my motor control and seeing how smal rotation steps i can make while still be very acurate. 

As seen in the picture below i use two Medium motors to run simultaneously along the same axis on a gearwrack (Y-axis). If i turn the motors more than 10 degrees they run and brake in the interval of +- 2 degree, depending on rotation speed. The problem is if I turn the mortars in steps under 10 degress the start to run unsynchronized. Approximately half of the steps are in turns and half of the steps are "together". I am looking for a method to run them completly synchronized in small steps (between 1 and 6 degress). I have look around in  different Issues with out any luck, so I am now trying to make my owen issue. 

I am looking for gide/help/ tips. 
Thanks for your help. 

**The code i am using now is:**
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor, InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile

ev3 = EV3Brick()

Ya_motor = Motor(Port.A)
Yb_motor = Motor(Port.B)
X_motor = Motor(Port.C)
def yaksecontrol(speed, ang, motor_1, motor_2):
    motor_1.run_angle(speed,ang,Stop.HOLD,False)
    motor_2.run_angle(speed,-ang,Stop.HOLD)
ev3.speaker.beep() 
ang_s=400
for i in range(208):
    yaksecontrol(ang_s, 4, Ya_motor, Yb_motor)
    print(round((abs(Ya_motor.angle())+abs(Yb_motor.angle()))/2))

for i in range(208):
    yaksecontrol(ang_s, -4, Ya_motor, Yb_motor)
    print(round((abs(Ya_motor.angle())+abs(Yb_motor.angle()))/2,1))

[image]

dlech, 2021-01-20:
You could try adding gears so that the motor position doesn't have to be as accurate. The motors have a few degrees of play, so it is not really possible to accurately control them within 1 degree. (And there is a gear ratio parameter for the motors, so you don't have to change the rest of your code.)

MrNoZzz, 2021-01-26:
Hi dlech.
I added a geaing 8:24 teeth and i runs pretty smooth now. Thanks for the help.