ev3dev/ev3dev #1464: EV3 Color display - turn the display in configuration

cho934, 2021-01-22:
Hi,

I connected the adafruit 1.8" tft to the EV3, then I saw in this video
https://www.youtube.com/watch?v=gPNJC5Uz9HY
that it seems possible to change the EV3 Board configuration to turn the display, but where is it ?

I already activated the screen editing the env.txt in the SD root. Is it in this file that we can turn the display ?

BR,

dlech, 2021-01-22:
If you are building your own kernel, it is in the device tree. If not, you can rotate it 180 degrees by using a different device tree file as discussed in #962.

cho934, 2021-01-23:
Thanks. 
So I confirmed that I 'm not building my own kernel. 

I got it!
=> for inverted position : edit the env.txt with (commented by default in this file);
fdtfile=da850-lego-ev3-adafruit18.dtb

=> for normal position : edit the env.txt with:
fdtfile=da850-lego-ev3-color.dtb

Thanks.