ev3dev/ev3dev #1465: set_dc_duty_cycle_sp fails to work first time

zacn, 2021-01-28:
The set_dc_duty_cycle_sp function in C fails to work the first time but works thereafter.

The code is a modified version of the dc motor example from: http://in4lio.github.io/ev3dev-c/
It was compiled via docker apparently using: ev3dev/debian-stretch-cross
It was run on an EV3 brick using: ev3dev-stretch-ev3-generic-2020-04-10

Code:

#include <string.h>
#include <stdio.h>
#include <unistd.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>

#include "ev3.h"
#include "ev3_dc.h"
#include "ev3_port.h"
#include "ev3_tacho.h"
#include "ev3_sensor.h"

int main(void){

  uint8_t sn, sn_port;
  char    charbuf[256];

  ev3_init();
  ev3_port_init();

  sn_port = ev3_search_port( OUTPUT_B, EXT_PORT__NONE_ );
  set_port_mode_inx( sn_port, EV3_OUTPUT_DC_MOTOR );

  ev3_dc_init();
  
  int  dutycyclesp;

  if ( ev3_search_dc_plugged_in( OUTPUT_B, EXT_PORT__NONE_, &sn, 0 )) {
    set_dc_duty_cycle_sp( sn, 100 );
    get_dc_duty_cycle_sp(sn, &dutycyclesp);
    printf("duty_cycle_sp is %i \n", dutycyclesp);
    set_dc_command_inx( sn, DC_RUN_FOREVER );

    usleep(2500000);
    get_dc_state(sn, charbuf, sizeof(charbuf));
    printf("dc motor state is %s \n", charbuf);
    usleep(2500000);
    
    set_dc_command_inx( sn, DC_STOP );
  }

  if ( ev3_search_dc_plugged_in( OUTPUT_B, EXT_PORT__NONE_, &sn, 0 )) {
    set_dc_duty_cycle_sp( sn, 100 );
    get_dc_duty_cycle_sp(sn, &dutycyclesp);
    printf("duty_cycle_sp is %i \n", dutycyclesp);
    set_dc_command_inx( sn, DC_RUN_FOREVER );

    usleep(2500000);
    get_dc_state(sn, charbuf, sizeof(charbuf));
    printf("dc motor state is %s \n", charbuf);
    usleep(2500000);
    
    set_dc_command_inx( sn, DC_STOP );
  }

  set_port_mode_inx( sn_port, EV3_OUTPUT_AUTO );

  ev3_uninit();

}

End-of-Code

With an RCX motor attached to port OUT_B on the EV3 brick running the program produces:

duty_cycle_sp is 0 
dc motor state is  
duty_cycle_sp is 100 
dc motor state is running 

There is nothing obviously wrong with the hardware. Running the low level commands:

echo 100 > /sys/class/dc-motor/motor0/duty_cycle_sp
echo run-forever > /sys/class/dc-motor/motor0/command
echo stop > /sys/class/dc-motor/motor0/command

on the EV3 brick via SSH the motor runs first time.

dlech, 2021-01-28:
Try adding a delay after `set_port_mode_inx` to give the kernel time to configure the new device.

zacn, 2021-01-29:
Thank you David. I delay of between 0.1 and 1 second after the set_port_mode_inx function solves the problem. A delay of between 1 and 2 seconds works when setting up two dc motors.

I will close the issue as solved.

closed 2021-01-29 by zacn