ev3dev/ev3dev #1467: add Valgrind in the next release

cho934, 2021-02-07:
Hi,
We are working a lot with C++, and I'm looking for some debugger.

Would it be possible to add valgrind in the next release ?
https://valgrind.org/

BR,
Christophe

dlech, 2021-02-07:
Ev3dev already includes `gdbserver` for remote debugging. My experience with trying to run debuggers completely on the EV3 is that it just doesn't have enough CPU and RAM to handle it. But you should already be able to install the valgrind debian package if you want to try for yourself.

cho934, 2021-02-07:
I unsuccessfully tried, and received this error message 

`checking host system type... armv5tejl-unknown-linux-gnueabi` 
`checking for a supported CPU... no (armv5tejl)` 
`configure:error: Unsupported host architecture. Sorry` 

ok will check with gdbserver ;)

closed 2021-02-12 by cho934