ev3dev/ev3dev #1468: "fast" I2C at 100kHz or 400kHz ?

cho934, 2021-02-12:
Hi,

I saw the issue #506, but I don't understand the conclusion. 
Ok to modify the pullup resistor, or even to use a i2c level shifter. But what to do to increase the speed of the i2c on the sensor port 1 or 2 or 3 or 4 ?

BR,
Cho

dlech, 2021-02-12:
You will probably have to build your own kernel for this ([link](https://github.com/ev3dev/ev3dev-buildscripts/)). The timer configuration for the I2C clock is [here](https://github.com/ev3dev/ev3-kernel/blob/ad1f6c8727bce29ec400c9385f7f72ee2c0e59fc/arch/arm/mach-davinci/time.c#L210-L217). The timer is currently set to 20kHz. It triggers both the rising and falling edge of the I2C clock, so the resulting clock is 10kHz. So, to get 100kHz, change 50 to 5. This will change the speed of all ports. More work would need to be done to make them individually configurable.

cho934, 2021-03-02:
Thanks dlech, I've just build it successfully for ev3-stretch.

Please, could you remind me how to add/build the rtl8812au-modules (like #1021 but for ev3-stretch), because it seems not included (but it is in the last stretch release ) ?

dlech, 2021-03-02:
https://github.com/ev3dev/ev3dev/issues/1021#issuecomment-358367313

I've only built them for ev3dev releases. I don't know if there is an easier way to build a one-off version.

cho934, 2021-03-03:
> #1021 (comment)

In fact, i would like to add it in the ev3dev like in the last release.
I don't understand why I lost it with my custom kernel build, i've just
follow the ev3dev-stretch tuto?

dlech, 2021-03-03:
> I don't understand why I lost it with my custom kernel build

Because the versions don't match. The new kernel is `4.14.117-ev3dev-2.3.5-pmx-ev3+` but the old kernel is `4.14.117-ev3dev-2.3.5-ev3`

dlech, 2021-03-03:
> What I need to do to have the same generic kernel you built ?

This is what I use to build ev3dev kernel releases, including the extra wifi drivers: https://github.com/ev3dev/ev3dev-kpkg

cho934, 2021-03-03:
ok I understand that this is the huge build for all environments.
but is there an easy way to just add the modules (extra wifi drivers) to the ev3dev-buildscripts (and build only for ev3)?

dlech, 2021-03-03:
I don't know how to add extra stuff to the buildscripts without a bunch of work. By setting environment variables, you can use ev3dev-kpkg to build just for EV3. 

Or maybe this helps: https://github.com/ev3dev/ev3dev/issues/1442#issuecomment-757471893

I'm not sure how they compiled the wifi driver. Maybe on the EV3?

cho934, 2021-03-03:
> I don't know how to add extra stuff to the buildscripts without a bunch of work. By setting environment variables, you can use ev3dev-kpkg to build just for EV3.

Yes, I will add the I2c modification and that's the goal ;)
What I don't understand is that the build-release seems to add a git tag that already exist. 
So I need to remove both tests (from line 87 to 94) ? Right ?
Another question, when using this method, how to configure the kernel with ./menuconfig ?

Another idea: I can retrieve the ko files from your release build
`robot@ev3dev:~/pmx$ ls /lib/modules/4.14.117-ev3dev-2.3.5-ev3/drivers/net/wireless/
8188eu.ko  rtl8812au.ko`
but it doesn't work for rtl8812au
`robot@ev3dev:~/pmx$ sudo insmod drivers/net/wireless/8188eu.ko 
robot@ev3dev:~/pmx$ sudo insmod drivers/net/wireless/rtl8812au.ko 
insmod: ERROR: could not insert module drivers/net/wireless/rtl8812au.ko: Unknown symbol in module
`

dlech, 2021-03-03:
It looks like you can pass `test` as a command line argument to `build-release` to skip the git tag and package signing.

For menuconfig, you should be able to run `make ARCH=arm menuconfig`.

cho934, 2021-05-25:
Hi,

For information, I made the modification into ```time.c``` to increase the speed of I2c to 100kHz
I built successfully with ```FLAVOR=ev3 ./build-release test``` the kernel and the additional wifi drivers (using the ev3dev-stretch version https://github.com/ev3dev/ev3dev-kpkg/tree/ev3dev-stretch )

You can add ```debhelper``` in the readme in the *Build depends* (I used ubuntu 18.04 to build)
```sudo apt install debhelper```

then also these two commands ```FLAVOR=ev3 ./build-release test``` and ```make ARCH=arm menuconfig``` in the ev3-kernel directory.

In my first tests, I2c at this speed works good. I will let you know if I have issue in the future.

Cheers