ev3dev/ev3dev #1469: Importing modules problem

Maduser1, 2021-02-13:
**Question**
Hello, 
I am new to python/ev3dev and I am tyring to set everything up before I start working with bricks. For now I did everything what was written here: https://github.com/ev3dev/vscode-hello-python
but I encountered some problems in regards to importing modules/code completion:

    Here: https://sites.google.com/site/ev3devpython/setting-up-vs-code
at the very bottom of the page there is example code, which begins with "from ev3dev.ev3 import *" but after execution of this line, I got error: "ModuleNotFoundError: No module named 'ev3dev' ". I entered 'lib' folder in my .venv folder and noticed that indeed there is no such module so I assumed that this instruction is not up to date. However in the 'lib' I found ev3dev2, so I tried: "from ev3dev2 import *" that caused no error, but it clearly did nothing as code completion did not work. In the ev3dev folder there were some modules, so I tried import them individually. I imported ev3dev.auto which finally made code completion work. However my simple question is: why  "from ev3dev2 import *" does not work, and if importing 'ev3dev.auto' provides all the available functionality? (I mean if code completion will work with all available functions and they will work fine?) If not, I will be grateful for explanation how this should be done (is there a way to import all available modules at once?), because I feel that I did something stupid and I made it work by accident.

I apologize if this question is really dumb, and I will much appreciate casting some light on that matter, because I prefer to know how things work :)

**Desktop (please complete the following information):**
 - Win 10
 -python 3.9.1

dlech, 2021-02-13:
This will happen if you run the code on your computer instead of on the robot. Made sure you are using the ev3dev VS Code extension and use it to download and run the code on the EV3.

Maduser1, 2021-02-14:
Thank you very much @dlech. I will proceed to work with bricks then :D

closed 2021-02-15 by Maduser1