ev3dev/ev3dev #1471: ev3dev-stretch buildscript problem!

we8809, 2021-02-20:
I have a problem.

Build was success. 

And i run this code ->
~/work/ev3dev-buildscripts $ ./build-kernel bindeb-pkg KDEB_PKGVERSION=1
...
<lots-of-build-output>
...
~/work/ev3dev-buildscripts $ ls ./build-area/*.deb
./build-area/linux-headers-3.16.7-ckt9-5-ev3dev-your-name-ev3_1_armel.deb
./build-area/linux-image-3.16.7-ckt9-5-ev3dev-your-name-ev3_1_armel.deb
./build-area/linux-libc-dev_1_armel.deb

Finally, I get a ev3dev-script-debian package.
and then, 
1) transfer linux-image* file to ev3 board 
2) sudo dpkg --install linux-image*.deb
3) sudo reboot

but system is not boot 
what is the problem?
Am I missing any process?

dlech, 2021-02-20:
Are you using ev3dev-jessie or ev3dev-stretch?

we8809, 2021-02-20:
I'm using stretch version!

dlech, 2021-02-20:
It looks like you have built a kernel for ev3dev-jessie. Be sure to that you use the `ev3dev-stretch` branch of all of the git repositories.

we8809, 2021-02-20:
root@seunghwan-900X5N:/home/seunghwan/lego_ev3dev/ev3-kernel# git checkout
M	drivers/lego
Your branch is up to date with 'origin/ev3dev-stretch'.

Kernel branch is ev3dev-stretch for now.

Also ev3dev-buildscript branch is ev3dev-stretch too.

I'm so confused about that..

dlech, 2021-02-20:
I'm not sure how you got kernel version `3.16.7-ckt9-5` then. It should be `4.14.x`.

we8809, 2021-02-21:
Actually, I ran this command.

> ~/work/ev3dev-buildscripts $ ./build-kernel bindeb-pkg KDEB_PKGVERSION=1
...
<lots-of-build-output>
./build-area/linux-headers-4.11.117-ev3dev-2.3.5-seunghwan-ev3+_1_armel.deb
./build-area/linux-image-4.11.117-ev3dev-2.3.5-seunghwan-ev3+_1_armel.deb
./build-area/linux-libc-dev_1_armel.deb

I get above output.
And I transfer linux-image-4.11.117-ev3dev-2.3.5-seunghwan-ev3+_1_armel.deb file to ev3 board's /home/robot.
then I ran this code

> sudo dpkg --install linux-image-4.11.117-ev3dev-2.3.5-seunghwan-ev3+_1_armel.deb

Successfully installed and reboot ev3 system.
But, system was not booting....

dlech, 2021-02-21:
`4.11` still sounds like the wrong kernel version.

we8809, 2021-02-21:
Oh I'm sorry 
> linux-image-4.14.117-ev3dev-2.3.5-seunghwan-ev3+_1_armel.deb
This is real file name. 
kernel version is 4.14.117!

we8809, 2021-02-24:
Sir, 
kernel version is truly right.
and debian image coming out -> linux-image-**4.14.117**-ev3dev-2.3.5-seunghwan-ev3+_1_armel.deb 

and I transfer this file to ev3 board's /home/robot folder,
then I install it, rebooting...

but ev3 not boot..

Once the ev3 board had a stretch image that i've downloaded from ev3dev website in it.
, am i missing any part of the buildscript process? 

I really hope modify the ev3dev kernel ㅠㅠ

dlech, 2021-02-24:
It has always worked for me, but I have not tried it for a while. Are there any error messages on the screen of the EV3? Do you have a UART adapter that you could attach to input port 1 for debugging?

dlech, 2021-02-24:
The `+` in the file name tells me that you have modified the kernel already (or maybe typed `KDEB_PKGVERSION=1` incorrectly`) Try building without any modifications first.