ev3dev/ev3dev #1473: ActuonixL1250Motor is not being detected

sheenamelwani52, 2021-02-23:
I am trying to use ev3 actuonix L1250 linear motor with feedback control using ActuonixL1250Motor Class but it not being detected on EV3. 

When I run dmesg | tail, it detects as an ev3 Large Motor instead of linear. 
[image]

I also tried to run it with LargeMotor and Motor class but in that case, position and position_sp do not provide proper values. It remains 0 even after the motor is fully contracted or extended. And also count_per_m will not work since it is not detected as a linear motor.

**Code:**

from ev3dev2.motor import Motor, ActuonixL1250Motor, LargeMotor, OUTPUT_B, SpeedPercent

def main():
    m = ActuonixL1250Motor(OUTPUT_B)
    reset_console()
    set_cursor(OFF)
    set_font('Lat15-Terminus24x12')
    print ("Postion {}".format(m.position))
    print ("Tachos {}".format(m.position_sp))
    print ("count/rot {}".format(m.count_per_rot))
    #print ("count/m {}".format(m.count_per_m))
    
    m.on_for_seconds(SpeedPercent(-75), 1)

    time.sleep(5)
   
    reset_console()
    set_cursor(OFF)
    set_font('Lat15-Terminus24x12')

    print ("Postion{}".format(m.position))
    print ("Tachos {}".format(m.position_sp))
    print ("count/rot {}".format(m.count_per_rot))
   #print ("count/m {}".format(m.count_per_m))
    time.sleep(5)

dlech, 2021-02-23:
These motors can't be automatically detected. You will need to use a [LegoPort](https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/stable/ports.html#ev3dev2.port.LegoPort) object and call `set_device = 'act-l12-ev3-50'` to get the correct driver. You will probably need a delay after this before creating the `ActuonixL1250Motor` object.

dlech, 2021-02-23:
Alternately, you can use the device browser on the EV3 to manually configure the port.

sheenamelwani52, 2021-02-23:
Thank you for your quick response. I did the following edits as suggested

from ev3dev2 import list_devices
from ev3dev2.port import LegoPort

  p1 = LegoPort(OUTPUT_B)
  p1.set_device = 'act-l12-ev3-50'
  time.sleep(2)

However, it is still not resolved, my visual studio output
[image]

In dmesg | tail, I can see now it is detecting but also automatically removing?
[image]

dlech, 2021-02-23:
Might need to take the port out of `auto`

    p1.mode = 'tacho-motor'

sheenamelwani52, 2021-02-28:
Thanks, I tried that already but still gives the same error.

dlech, 2021-02-28:
BrickPi3 treats everything as an NXT motor IIRC.

sheenamelwani52, 2021-02-28:
Yes, that's right but can I configure the port by the same method as you have mentioned for ev3?