ev3dev/ev3dev #1474: cannot connect to RPI using wired connection via ssh

sheenamelwani52, 2021-03-01:
I am trying to connect to ev3dev on RPI3 (with brickPi3) first time using a wired ethernet connection via ssh. It says "could not find host". 
Also tried to ping ev3dev and ev3dev.local but same error again. 

[image]

Can you please guide regarding this?

**Desktop **
 - OS  Window 10 
 - Connection Type: Wired Ethernet
 - Software:  PuTTY, windows command prompt

**Robot**
 - Device:  BrickPi3

dlech, 2021-03-01:
Try using the ev3dev VS Code extension to make the initial connection.

xdsl, 2021-04-01:
You can connect by ip address