ev3dev/ev3dev #1475: how to communicate from python with the ev3?

moldan2005, 2021-03-16:
I have a raspberry pi 4 and an Lego Mindstorm EV3 with ev3dev on it, and I wondered to combine these together with USB cable for the fast communication. Now, I want the EV3 to make a beep sound, like this guy did but it doesn't work for me.
(http://mvidner.blogspot.com/2017/01/usb-communication-with-python-and-pyusb.html)

**the code:**
```
import sys
import usb.core

def ev3_write(command):

    VENDOR_LEGO = 0x0694       # lsusb
    PRODUCT_EV3 = 5
    CONFIGURATION_EV3 = 1       # 1-based
    INTERFACE_EV3 = 0           # 0-based
    SETTING_EV3 = 0             # 0-based
    ENDPOINT_EV3 = 1            # 0-based

    device = usb.core.find(idVendor=VENDOR_LEGO, idProduct=PRODUCT_EV3)
    if device is None:
        print("Is the brick connected and turned on?")
        sys.exit(1)

    if device.is_kernel_driver_active(INTERFACE_EV3):
        print("Detaching kernel driver")
        device.detach_kernel_driver(INTERFACE_EV3)

    configuration = device.get_active_configuration()

    interface = configuration[(INTERFACE_EV3, SETTING_EV3)]

    endpoint = interface[ENDPOINT_EV3]

    endpoint.write(command)

beep_command = '\x0F\x00\x01\x00\x80\x00\x00\x94\x01\x81\x02\x82\xE8\x03\x82\xE8\x03'
ev3_write(beep_command)
```
**After running the code , I've got this output, and error:**
```
Detaching kernel driver
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 236, in get_interface_and_endpoint
    return self._ep_info[endpoint_address]
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/_____python/USB/usb_v1.1.py", line 31, in <module>
    ev3_write(beep_command)
  File "/home/pi/_____python/USB/usb_v1.1.py", line 28, in ev3_write
    endpoint.write(command)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 408, in write
    return self.device.write(self, data, timeout)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 976, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 228, in setup_request
    intf, ep = self.get_interface_and_endpoint(device, endpoint_address)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/usb/core.py", line 244, in get_interface_and_endpoint
    raise ValueError('Invalid endpoint address ' + hex(endpoint_address))
ValueError: Invalid endpoint address 0x0
```

Can anyone halp me with this? Thank you in advance for your answers.
(this is my first stackoverflow question, and my english is not too good, so sorry for that)
The exact same question is on stackoverlow too
(https://stackoverflow.com/questions/66654786/how-to-make-beep-sound-from-the-rapberry-pi-on-the-ev3-with-ev3dev-via-usb-conne)

dlech, 2021-03-16:
I think they are using the official LEGO firmware, not ev3dev.

moldan2005, 2021-03-16:
Thanks for the idea, I tried it out, and I didn't have any error, but he didn't make the beep sound. Any idea?

dlech, 2021-03-16:
Maybe the volume is turned down? Or the command was not correct?

moldan2005, 2021-03-16:
The volume is on, then the command shouldn't be right.

closed 2021-03-18 by moldan2005