ev3dev/ev3dev #1478: Real-time data loggin from brick to PC

Maduser1, 2021-03-25:
**Question**
Hello, in my actual project I need to be able to save data from sensors to .csv file. That is easy enough and I wrote simple function in Python, which successfully saves data logs from the sensors on the run. The issue is that I need the .csv file to be saved not in the brick's memory but on my PC directly.

My function to save data from sensors looks like this:

```
 def saveData():
    with open('data.csv', 'a', newline='') as csvfile:
        fieldnames = ['Time', 'Color']
        writer = csv.DictWriter(csvfile, fieldnames=fieldnames)
        writer.writerow({'Time': now_datetime, 'Color': checker})
    return
```

**WHAT I TRIED**

- to set path to my PC, like 'C:/user/desktop.data.csv but as expected, that resulted with error: _FileNotFoundError: [Errno 2] No such file or directory_  
- tried the same with RPyC but with the same results. I am not sure if I set everything up the correct way, but it seems I did.

Is there a way to achieve something like this whatsoever? I just want the .csv file to save directly on my PC, so that I do not have to download it from the brick. If someone know solution to this, please put me on the right track regarding this matter. 

**Desktop:**
 - Win 10
 - Connection Type: Bluetooth network connection 
 - Software: VSC

**Robot:**
 - Device: EV3

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=7
BOARD_INFO_SERIAL_NUM=001653820B7E
BOARD_INFO_TYPE=main
```

dlech, 2021-03-26:
Would it work to copy the file from the EV3 to the PC later using WinSCP or Filezilla?

Maduser1, 2021-03-26:
Hello, thank you for responding. 

Unfortunately it is crucial for me to have this file being saved on PC directly, because this data is necessary for further usage while the program is still on the run. So what I need, is being able to have look at the data as it is being written, with for example notepad++. I managed to achieve this with combination of EV3 mindstorms, small basic and daisy chaining, but I would rather use Python for better stability, readability and other factors. However, if what I want to achieve is impossible or very challenging, I will then stay with my current solution. 

Regards.