ev3dev/ev3dev #1479: How do I read the buttons in C

hpvpp, 2021-03-31:
**Question**
How do I read the buttons in C

**Background**
[test-keys-2.zip](https://github.com/ev3dev/ev3dev/files/6234214/test-keys-2.zip)

I am trying to port GNU Chess 5.08 to the EV3.
My plan is, firstly, to use the buttons tonavigate a set of menus and , secondly, to use the buttons to provide input (chess moves) to the program.

What I have manage to achieve so far is that when I push the buttons on the EV3 I get scan codes and not the codes in de doco on http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/ev3.html#buttons.  Running my latest attempt (see attached) through the file manager of brickman, gives this result

1 - clears screen
2 - displays my startup message and waits for input
3 - pressing any of the four arrow keys prints the scan code
4 - pressing <enter> displays my text that goes with the arrow
5 - pressing <back> has no visible effect
6 - pressing <enter> again exits the program

**Desktop:**
 - OS and version: Ubuntu 20.04.2 LTS
 - Connection Type: USB
 - Software: C

**Robot:**
 - Device: EV3

[test-keys-2.zip](https://github.com/ev3dev/ev3dev/files/6234200/test-keys-2.zip)
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653536ABB
BOARD_INFO_TYPE=main
```

dlech, 2021-03-31:
Why the older version? Is the [debian packaged version](https://packages.debian.org/stretch/gnuchess) to slow?

The buttons on the EV3 act as keyboard keys. Up, down, left, right arrow keys, enter and backspace. So you can read them from stdin if the program is running on a virtual console on the EV3 or you can use the Linux event input subsystem (libevdev, libinput or similar).

dlech, 2021-03-31:
Here is the unzipped program.

```C
/* test-keys-2.c */

#include <stddef.h>
#include <stdio.h>
#include <string.h>

#define MAX_STRING_SIZE 128

int main(void) {

  char input[MAX_STRING_SIZE] = {0};

  printf("test-keys-2\n");
  
  for (;;) {
    fgets(input, sizeof(input), stdin);

    if (input[0] == '\33') {
      switch (input[2]) {
      case 'A':
        printf("^^^^^UP\n");
        break;
      case 'B':
        printf("vvvvvDOWN\n");
        break;
      case 'C':
        printf(">>>>>    RIGHT\n");
        break;
      case 'D':
        printf("<<<<<LEFT\n");
        break;
      default:
        printf("UNKNOWN: %s\n", input);
        break;
      }
    } else if (input[0] == '\n') {
      printf("ENTER\n");
      break;
    } else {
      printf("%zu characters were read.\n", strlen(input));
      printf("input was: %s\n", input);
      break;
    }
  }
  
  return 0;
}
/* test-keys-2.c */

```

dlech, 2021-03-31:
The ncurses library is also good for getting keyboard input.

xdsl, 2021-04-01:
Using an external keyboard or brick buttons?
If you are using a brick button, this example for you: https://www.ev3dev.org/docs/tutorials/using-ev3-buttons/
Just translate it from Python to C.

hpvpp, 2021-04-02:
ncurses works well for me (thank you), but KEY_BACKSPACE is captured by brickman/brickrun.
I _do_ need control of that key.
How do I keep control of the KEY_BACKSPACE ?

dlech, 2021-04-02:
You can add `delay=1` to `/etc/brickrun.conf` to defer it.

hpvpp, 2021-04-02:
works well, thank you