ev3dev/ev3dev #1480: gsettings

hpvpp, 2021-04-03:
**Question**

How do increase text size?

I found that I can increase the text size by using the command 'gsettings' and that works well on my Ubuntu box, but on the brick it  fails as follows:
robot@ev3dev:~/src$ gsettings set org.gnome.desktop.interface text-scaling-factor 2
-bash: gsettings: command not found

apt list reports:
robot@ev3dev:~/src$ apt list --installed libglib*
Listing... Done
libglib2.0-0/oldstable,now 2.50.3-2+deb9u2 armel [installed,automatic]

**Background Information**

I am using ncurses, because it appears that the GRX library does not have support for C as I found out here http://docs.ev3dev.org/projects/grx/en/latest/

**Robot (please complete the following information):**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653536ABB
BOARD_INFO_TYPE=main
```

dlech, 2021-04-03:
The C API for GRX is here: http://docs.ev3dev.org/projects/grx/en/latest/c-api/index.html

FYI, GRX is graphics mode only (so can't be used in conjunction with ncurses) and is known to have poor performance when using fonts.

If you are using ncurses instead, then you need to set the console font, which can be done with the `setfont` command or equivalent ioctls. Available fonts can be found in `/usr/share/consolefonts/`.

You may find some useful info here: https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/stable/console.html

hpvpp, 2021-04-04:
Thanks for the link to GRX, but I prefer to use ncurses, being the easier option and more at my level of expertise.  

The `setfont` command returns *Couldn't get a file descriptor referring to the console*, both on my Ubuntu desktop and on the brick.  I have googled this, but can't figure it out.  The python example didn't help me, sorry.

Concerning the `gsettings` command, I installed `sudo apt install libglib2.0-0-dbg` on the brick (as per https://command-not-found.com/gsettings), but I still get the response *command not found*.  

I would very much prefer the `gsettings` command, because that does not require knowledge of font availability.

dlech, 2021-04-04:
ev3dev doesn't use gsettings for anything (it is not running any desktop software), so I don't see how that could help you.

`setfont` must be run in a virtual console on the EV3. In other words, it won't work unless you use `brickrun` to start your program.

hpvpp, 2021-04-04:
**It works!**  Thank you so much David.

Not being able to get `setfont` to work in a terminal window on my desktop, I assumed it also wouldn't work on the EV3.  Shows how little I understand about Linux :-(