ev3dev/ev3dev #1482: UART communication between ev3 (input port1) and arduino is not working

amreo, 2021-04-09:
**What are you trying to do?**
I'm trying to use UART for communicating between arduino and ev3, and print some receiving bytes from the arduino
```
#!/usr/bin/env pybricks-micropython
from pybricks.parameters import Port, Color, Button, Direction, Stop
from pybricks.iodevices import DCMotor, UARTDevice, LUMPDevice, I2CDevice
x = UARTDevice(Port.S1, 9600)
print(x.read_all())
```

**What did you expect to happen**
This simple micropython script prints something.

**What actually happened**
The script prints `b''`

**Source Code**
Here is the program I am trying to run.

(ev3)
```
from time import sleep
brickrun -r -- pybricks-micropython
from pybricks.parameters import Port, Color, Button, Direction, Stop
from pybricks.iodevices import DCMotor, UARTDevice, LUMPDevice, I2CDevice
x = UARTDevice(Port.S1, 9600)
print(x.read_all())
```
(arduino uno)
```
#include <SoftwareSerial.h>

SoftwareSerial Serial1(12, 11); // RX, TX

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  Serial1.begin(9600);

  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
}

void loop() {
  Serial.print("A");
  Serial1.print("A");
}
```

**Arduino UNO (please complete the following information):**
 - OS and version: none
 - Connection Type: UART (other-uart)
 - Software: arduino example code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev-stretch-ev3-generic-2020-04-10, 4.14.117-ev3dev-2.3.5-ev3

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165352B390
BOARD_INFO_TYPE=main
```

**Background Information**
I am trying to create a UART(and not i2c that works) communication between ev3 and arduino for experimental purpose...
I have also tried to use the LUMP protocol (https://lejosnews.wordpress.com/2014/06/03/arduino-uart-sensors/) with no success

**wire links**
Pin#1(ev3) (white) --- GND(arduino)
Pin#2 (ev3) (black) NC
Pin#3 (ev3) (red) --- GND(arduino) --- Pin#1(ev3) (white)
Pin#4 (ev3) (green) --- 5V (arduino)
Pin#5 (ev3) (yellow) --- D12 (arduino)
Pin#6 (ev3) (blue) --- D11 (arduino)

dlech, 2021-04-09:
`read_all()` actually means "read what is currently in the buffer, if anything".  Creating the `UARTDevice` probably clears the buffer. So there probably isn't enough time to receive anything new before printing.

Try `x.read(1)` instead to block until it actually receives one byte.

> Pin#4 (ev3) (green) --- 5V (arduino)

Is this actually powering the Arduino? If not, don't connect it.

amreo, 2021-04-09:
> > Pin#4 (ev3) (green) --- 5V (arduino)
> 
> Is this actually powering the Arduino? If not, don't connect it.

No, Arduino is powered through USB, so I have removed the wire.

> Try `x.read(1)` instead to block until it actually receives one byte.

Done, and return `b'\x05'` insted of `b'A'` or `b'\x41`

dlech, 2021-04-09:
What does it do if you change it to this?

```python
while True:
    print(x.read(1))
```

amreo, 2021-04-09:
Print many b'A'... What?

amreo, 2021-04-09:
@dlech after I have rebooted EV3 (or better I have wrongly shortcuted vcc and gnd) I continue to get b'\x05' instead of b'A'...

dlech, 2021-04-09:
Maybe a timing issue with the SoftwareSerial on the Arduino? This is the point where I would get out the oscilloscope or logic analyzer to see what is going on.

amreo, 2021-04-09:
I don't have a oscilloscope or logic analyzer. The output isn't always determistic. I have tried to use the hardware serial but the output isn't the right output and determistic. I suspect is a hardware/connection problem because moving the wire change the output...

Oxymore-coder, 2022-12-13:
A little late, but, ry unplugging the ground wires. They are not required for UART, and having them messes things up on my part.

amreo, 2022-12-19:
I have tried to unplug the ground wires, but the issue is the same

I have tried to analyze the signal with my (recent) oscilloscope. This is the waveform of D11 pin (with the arduino GND) and with white/red pin attached to ground.

[image]
[image]

amreo, 2022-12-22:
> Maybe using SoftwareSerial instead of builtin Serials is somehow messing it up, maybe with different UART configurations, so try `Serial`, `Serial1` or any other like in my code

Apparently it seems to work (but not using Serial1 because Arduino UNO doesnt't have it)

amreo, 2022-12-22:
But it seems unstable: with 57600 as baudrate and without the delay, it reads '\t' (and serial hw), even if in the doc there is written

> The UART communication functions as asynchronous communication,
supporting from 2400 bit/s to 460 Kbit/s on port 1 & 2, while ports 3 & 4 supports up to 230 Kbit/s.

(Hardware developer kit pdf)

Oxymore-coder, 2022-12-25:
That could be a limitation on the arduino's side. I rarely go above 9600 with devices that aren't my laptop, that makes it safe, and with short messages, it's not a problem. Don't go too high up and you should be fine