ev3dev/ev3dev #1483: Controling motors by keyboard

comand100vip, 2021-04-21:
**Question**
Is it possible to control motors by keybord?

**Background Information**
I am using ev3dev and python
I am trying to control car because I want to build a robot that go forwar then I hold 'W' or turn right then I click 'D'

**Screenshots**

**Desktop (please complete the following information):**
 - OS and version: Windows 8.1 (I don't knwo the build)
 - Connection Type: Bluetooht
 - Software: PuTTY

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]
```
<!-- Copy everything between these lines -->
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653513D46
BOARD_INFO_TYPE=main
```
<!-- Copy everything between these lines -->
```

dlech, 2021-04-21:
Will the keyboard be connected to the EV3 with Bluetooth?

comand100vip, 2021-04-21:
No, it is connected by cable to my laptop and laptop is connecting by blotooth to ev3 using PuTTy
I want to capture button by python script

dlech, 2021-04-21:
If you run your python program with `brickrun -r` in the PuTTY terminal, it will redirect stdio from the program to the terminal. They you can use `sys.stdin` in the Python program to read keyboard input. However, this won't be able to detect separate press and release events of the key. To do that, you would need to make a separate program that runs on your computer to read the key press events and then send that information to a program running on the EV3.

comand100vip, 2021-04-22:
How I can make that the program on my laptop can sent information to another on EV3 brick?
What librali I need to use?
How can I install pip on EV3 (I try a lot of ways)

dlech, 2021-04-22:
I don't recommend using `pip` on the EV3. Many popular Python libraries are packaged by Debian and can be installed with `apt`, e.g. `sudo apt update && sudo apt install python3-<name>`.

Sockets are part of the standard Python library, and don't need to be installed separately.

comand100vip, 2021-04-22:
Ok. this is all I need to know. Thanks

closed 2021-04-22 by comand100vip