ev3dev/ev3dev #1489: Install Lego drivers in Zynq board running Debian

deepakravibabu, 2021-05-19:
Hello,
I am working with EV3 lego sensor connected with Brickpi3. The Brickpi3 is stacked on the GPIO pins of Zynq board and the lego sensor connected with brickpi3. The Brickpi3 and Zynq board communicates with SPI (which is implemented now and wokring). But i want to install the lego drivers. The Zynq board is running Debian linux.
Could you please provide me some insights on how I can install the lego drivers?

**Accomplished tasks till now:** 
Currently, I have established spi communication on board
I downloaded the Brickpi3 repository. 
When I run the Test_Connected.py script in the pre-built Examples folder (Dexter/BrickPi3/Software/Python/Test_Connected.py) it gives a message ‘Brickpi3 connected and running’.
I have a EV3 Color sensor. When I run the example file (Dexter/BrickPi3/Software/Python/EV3_Color_Sensor_Color.py),  I am able to get the sensor values. 

**Issue:**
No lego drivers available under
_/sys/class/lego-sensor_
Could you please help me with how I can build lego drivers ?

Thanks for your time and consideration.

Kind Regards,
Deepak

dlech, 2021-05-19:
Its a bit out of date because no one was using it but DKMS is the way to go: https://github.com/ev3dev/lego-linux-drivers-dkms

deepakravibabu, 2021-05-19:
Please correct me if I am wrong.
As per my understanding, dkms package is used to directly connect lego sensor to board running linux (without brickpi3). But I want to use brickpi3.

dlech, 2021-05-19:
The brickpi3 drivers can be compiled using the same method. The package that I linked would have to be modified to add it.

deepakravibabu, 2021-05-28:
I cloned that repository and used make command to build the module but, there was no .ko file being generated.
[image]
Can you please tell where I am going wrong or provide some insights on how I can build the module.