ev3dev/ev3dev #1491: "sudo apt-get update" is stuck

dlech, 2021-05-27:
I am executing "sudo apt-get update"
robot@ev3dev:~$ sudo apt-get update
Ign:1 http://ftp.debian.org/debian stretch InRelease
Get:3 http://security.debian.org stretch/updates InRelease [53.0 kB]
Get:4 http://ftp.debian.org/debian stretch Release [118 kB]
Get:2 http://archive.ev3dev.org/debian stretch InRelease [8432 B]
Get:5 http://ftp.debian.org/debian stretch Release.gpg [2410 B]
Get:6 http://security.debian.org stretch/updates/contrib armel Packages [1524 B]
Get:7 http://security.debian.org stretch/updates/non-free armel Packages [5320 B]
Get:8 http://security.debian.org stretch/updates/main armel Packages [658 kB]
Get:9 http://archive.ev3dev.org/debian stretch/main armel Packages [82.7 kB]
Get:10 http://ftp.debian.org/debian stretch/non-free armel Packages [51.3 kB]
Get:11 http://ftp.debian.org/debian stretch/contrib armel Packages [39.5 kB]
Get:12 http://ftp.debian.org/debian stretch/main armel Packages [6879 kB]
0% [Working]^C

It does stuck...( as you can see in the output from the command)
I tried 2 different routes to internet to avoid any blocking, with no success.
Removed the packages under /var/lib/apt/list/partials,with no success.

Hope you can support here , thanks in advanced.

_Originally posted by @balharash in https://github.com/ev3dev/ev3dev/issues/873#issuecomment-849944020_

dlech, 2021-05-27:
`apt` is really slow on the EV3. How long did you wait?

balharash, 2021-05-27:
Around 15 min , but i can run it know the complete night.
I will post an update tomorrow morning, thank you for the quick response.

balharash, 2021-05-28:
Unfortunately, same behavior:

Ign:1 http://ftp.debian.org/debian stretch InRelease
Get:2 http://security.debian.org stretch/updates InRelease [53.0 kB]
Get:3 http://ftp.debian.org/debian stretch Release [118 kB]
Get:4 http://ftp.debian.org/debian stretch Release.gpg [2410 B]
Get:5 http://archive.ev3dev.org/debian stretch InRelease [8432 B]
Get:6 http://security.debian.org stretch/updates/non-free armel Packages [5320 B]
Get:7 http://security.debian.org stretch/updates/main armel Packages [658 kB]
Get:8 http://security.debian.org stretch/updates/contrib armel Packages [1524 B]
Get:9 http://ftp.debian.org/debian stretch/non-free armel Packages [51.3 kB]
Get:10 http://ftp.debian.org/debian stretch/contrib armel Packages [39.5 kB]
Get:11 http://ftp.debian.org/debian stretch/main armel Packages [6879 kB]
Get:12 http://archive.ev3dev.org/debian stretch/main armel Packages [82.7 kB]
0% [Working]

balharash, 2021-05-28:
I was able to solve the issue with:
sudo rm /etc/apt/source.list.d/* (backup before)