ev3dev/ev3dev #1492: ImportError: no module named 'multiprocessing.managers'

GDvW, 2021-05-28:
Python cannot find multiprocessing.managers

Steps to reproduce the behavior:
1. add 'from multiprocessing.managers import BaseManager' to main.py
2. run file
3. get error:
Traceback (most recent call last):
  File "/home/robot/**project**/main.py", line **, in <module>
ImportError: no module named 'multiprocessing.managers'

Used this program
'''#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile

from multiprocessing.managers import BaseManager

# Create your objects here.
ev3 = EV3Brick()

# Write your program here.
ev3.speaker.beep()'''

I expected no errors and a beep.

[image]

**Desktop (please complete the following information):**
 - OS: Window 10
 - Connection Type: USB
 - Other software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev2.0

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534B7A9B
BOARD_INFO_TYPE=main
```

dlech, 2021-05-28:
Pybricks uses MicroPython, not "regular" Python, so most Python libraries are not usable. What do you want to use the `multiprocessing` library for?

GDvW, 2021-05-29:
I want to make a car with a remote control, with processes for each function. I made my own class which I share between the different processes to share the data. However, just importing the multiprocessing library doesn't throw an error.
[image]

dlech, 2021-05-29:
You could use [ev3dev-lang-python](https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/ev3dev-stretch/?badge=ev3dev-stretch) to get Python 3.5.

Or in MicroPython you could use a different way to run code in parallel. My personal preference is to [use generators as coroutines](https://gist.github.com/dlech/066cc9593eecf88d044963a0a374f65e).

GDvW, 2021-05-31:
How do you use generators as coroutines? It was not clear for me out of the example.