ev3dev/ev3dev #1493: SSH to ev3

saranggoel, 2021-05-29:
**What are you trying to do?**

I am trying to ssh to my ev3 (with computer A - Windows) while using wifi over usb which is connected to another computer B (Linux).

**What did you expect to happen**

I should have been able to successfully ssh to the ev3.

**What actually happened**

I can ssh through computer B but not computer A. All putty says is the host is not found.

R2D2Hud, 2021-09-15:
Are the computers on the same network?

r0bis, 2022-05-29:
What I would do in such a case - check the ip address / netmask on all three computers (i.e. see that they are on the same subnet), then I would try pinging the ip address of ev3 from both computers A and B. If OK, I would try ssh to the ip address, that should work from both computers A and B.