ev3dev/ev3dev #1495: sudo apt-get update and sudo apt update not working

RebelOfDeath, 2021-06-04:
I am fairly new to this ev3dev thing, because I have received this as a project from school, but when I wanted to update apt and  apt-get it showed the following error

sudo apt-get update
Err:1 http://security.debian.org stretch/updates InRelease                                                    
  Could not resolve 'security.debian.org'
Err:2 http://httpredir.debian.org/debian stretch InRelease                                                    
  Could not resolve 'httpredir.debian.org'
Err:3 http://archive.ev3dev.org/debian stretch InRelease                  
  Could not resolve 'archive.ev3dev.org'
Reading package lists... Done        
W: Failed to fetch http://httpredir.debian.org/debian/dists/stretch/InRelease  Could not resolve 'httpredir.debian.org'
W: Failed to fetch http://security.debian.org/dists/stretch/updates/InRelease  Could not resolve 'security.debian.org'
W: Failed to fetch http://archive.ev3dev.org/debian/dists/stretch/InRelease  Could not resolve 'archive.ev3dev.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.

sudo apt update
Err:1 http://httpredir.debian.org/debian stretch InRelease                                                    
  Could not resolve 'httpredir.debian.org'
Err:2 http://security.debian.org stretch/updates InRelease                                                    
  Could not resolve 'security.debian.org'
Err:3 http://archive.ev3dev.org/debian stretch InRelease                 
  Could not resolve 'archive.ev3dev.org'
Reading package lists... Done        
Building dependency tree       
Reading state information... Done
All packages are up to date.
W: Failed to fetch http://httpredir.debian.org/debian/dists/stretch/InRelease  Could not resolve 'httpredir.debian.org'
W: Failed to fetch http://security.debian.org/dists/stretch/updates/InRelease  Could not resolve 'security.debian.org'
W: Failed to fetch http://archive.ev3dev.org/debian/dists/stretch/InRelease  Could not resolve 'archive.ev3dev.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.

I genuinely have no idea what went wrong and really need help. 

to also give some information about how the ev3 is connected to the internet, I am using the  ev3dev extension in vscode and also using putty. I am using a micro usb cable as an ethernet connection for the device and I can confirm that  it is definitely connected to the internet.

dlech, 2021-06-04:
Your EV3 must be connected to the internet for this to work. See https://www.ev3dev.org/docs/networking/

RebelOfDeath, 2021-06-05:
I have set up the connection exactly the way that it should be according to the tutorial, and I have also tried many solutions that you had offered to others with other issue tags, but my problem still remains that when I connect the brick to my laptop using the method with USB it only shows connected and not online.

dlech, 2021-06-05:
Do you have any security software that could be blocking the connection?

RebelOfDeath, 2021-06-05:
no, and to give you some more information I am running Ubuntu; I have tried both the USB method and the Bluetooth method which you have stated in previous issues, but non of the seem to work. I can confirm that I have all the necessarily packets and still with all this done I only get the message connected and not online.

dlech, 2021-06-05:
Can you share screenshots of your network configuration, both in Ubuntu and on the EV3?

RebelOfDeath, 2021-06-05:
[image]
[image]
[image]

[image]
[image]

hey thank you for responding so quickly, are these enough?

dlech, 2021-06-05:
What is the IP address on the EV3?

RebelOfDeath, 2021-06-06:
that would be 10.42.0.108

dlech, 2021-06-06:
It sounds like everything on the EV3 is working right and the problem is on your computer. Maybe the ideas in #1264 can help?

RebelOfDeath, 2021-06-06:
I will check it out; I wish you the best with this project, and I once again thank you for your help

closed 2021-06-06 by RebelOfDeath