ev3dev/ev3dev #1496: UDP port

CarloFlamingo, 2021-06-12:
In my current project I am sending direct lego firmware commands via WIFI to an EV3 running the original lego firmware along the lines of the beautiful tutorial on "http://ev3directcommands.blogspot.com". However, I need to do the same on an EV3 running ev3dev. 
Concerning this, I have two questions, first of all the basic one: 
(i) is this possible in principle, i.e. are the direct commands recognized in the same way by the ev3dev os as by the lego firmware? 
(ii) And, given the answer to (i) is 'yes', can you help me with the following problem: I try to connect to the brick running ev3dev with the python script for wifi connection on "http://ev3directcommands.blogspot.com/2016/01/no-title-specified-page-table-border_94.html". In this script port 3015 is hardcoded for receiving UDP data from an EV3 brick running the original lego firmware, which works fine.  But running the ev3dev os I do not receive any UDP data on port 3015. I guess this is because the ev3dev firmware just uses a different port. But how can I find out which port is actually used? Or is a yet different method necessary to establish the wifi connection for sending direct commands?
Thanks in advance for your help
Carlo

dlech, 2021-06-13:
Ev3dev doesn't use direct commands. What is the bigger picture of what you are trying to accomplish?

CarloFlamingo, 2021-06-14:
Hello David,
thank you for your reply.
The idea is to have an EV3 playing a game on a mobile phone, utiliizing a desktop computer in the following way:
• The lego robot turns the phone and touches its screen 
• the screen of the phone is mirrored to a more powerful computer and
• this computer analyses the mirrored screen, computes orders for the next movements and sends them to the EV3 

We use
• the lego mindstorms set 31313 with no additional hardware
• an Apple phone SE and
• an notebook running MacOS,
the programming language is python.
The game to be played we have in mind is ‚doodle jump‘ - broadly because it needs not-too-fast real-time reactions.

The context of my question is the communication from the notebook to the EV3: How to inform the robot about the next movements to make? 

Direct commands appear to me to be a good solution, but definitely I am open to other ways. Actually, there are no problems with direct commands using the original EV3 os, but since ev3dev has become a part of the life of my children and me, I would love to find a solution with it.

Things I tried but could not get to work:
• an NFS share worked far too slow
• bluetooth connection: the pybluez port for MacOS seems to be incomplete - at least my efforts to transfer examples from linux to MacOS ended up with error messages indicating that features are not yet implemented in pybluez for the apple universe. 

Definitely I missed something, maybe something obvious. I’d highly appreciate any hint how to establish a communication between a computer running MacOS and the EV3 brick running ev3dev such that a micropython program on the brick can receive instructions in round about real-time.

Thanks in advance 
Carsten

dlech, 2021-06-14:
https://pybricks.com/ev3-micropython/messaging.html on ev3dev should be compatible with https://r.jander.me.uk/index.php/2019/11/25/ev3-mailboxes-in-python/ on macOS.