ev3dev/ev3dev #1497: cannot change an apt source

yangshunhuai, 2021-06-19:
**What are you trying to do?**

I am trying to change the apt source to [Tsinghua TUNA](https://mirrors.tuna.tsinghua.edu.cn/) because the official source is so slow.

**What did you expect to happen**

The source works properly.

**What actually happened**

It stuck at this:
```
robot@ev3dev:/etc/apt$ sudo apt update
Hit:1 http://archive.ev3dev.org/debian stretch InRelease
0% [Working]
```

**Source Code**

This is the source I was using:

```
deb http://archive.ev3dev.org/debian stretch main
deb https://mirrors.tuna.tsinghua.edu.cn/debian/ stretch main contrib non-free
deb https://mirrors.tuna.tsinghua.edu.cn/debian/ stretch-updates main contrib non-free
```

**Desktop (please complete the following information):**
 - OS and version: Windows 7 64-bit
 - Connection Type: Wi-Fi
 - Software: PuTTY

**Robot (please complete the following information):**
 - Device: LEGO MINDSTORMS EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534DE339
BOARD_INFO_TYPE=main

```

dlech, 2021-06-19:
`http://archive.ev3dev.org` doesn't have any mirrors that I know of.

There haven't been any updates recently anyway, so you could just comment that line out.

yangshunhuai, 2021-06-21:
After commenting that line out, it still got stuck.

```
Linux ev3dev 4.14.117-ev3dev-2.3.5-ev3 #1 PREEMPT Sat Mar 7 12:54:39 CST 2020 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
Last login: Mon Jun 21 10:49:24 2021 from 192.168.4.192
robot@ev3dev:~$ sudo apt update
[sudo] password for robot: 
0% [Working]
```

dlech, 2021-06-21:
The EV3 is very slow. How long did you wait?

yangshunhuai, 2021-06-26:
It didn't respond in 10 minutes. Is it normal?

dlech, 2021-06-26:
Usually it doesn't take longer than that. But if you have a very slow SD card and/or very slow Internet, it might take longer.

yangshunhuai, 2021-06-27:
That seems not normal.
I am using a Toshiba 2GB SD Card and a 200M broadband Internet connection.