ev3dev/ev3dev #1498: Use USB devices on the mini-USB port

spyrolg, 2021-06-22:
I noticed that devices such as physical USB-ethernet adapters and keyboards work fine on the USB-A port but don't work on the mini USB port. I would like to use the mini USB port to connect my USB-ethernet adapter.

I seems to me that this port is configured and statically claimed for the 'software' USB ethernet adapter (ethernet connection over USB). Would it be possible to reconfigure ev3dev to make this a 'regular' USB port?
For my use case it is not required that the 'software' USB ethernet adapter' is available.

I consider this a local change that I make to my specific ev3dev instance, I'm not requesting this as a feature for ev3dev.

**Desktop:**
 - Windows 10 Pro Version	10.0.19043 Build 19043
 - Connection Type: USB

**Robot :** EV3, Ev3Dev-Stretch

dlech, 2021-06-22:
The port is hard-wired as a peripheral-only port. It can't supply 5V, so it can't work as a host port. (Technically, it is possible to make it kind of work if you do some hardware hacking and some kernel hacking, but probably not in a standards compliant way.)

spyrolg, 2021-06-23:
That explains it perfectly, thank you for your quick answer.

closed 2021-06-23 by spyrolg