ev3dev/ev3dev #1501: Wifi "Not available"

ZonEmOToRY, 2021-07-02:
I recently bought a wireless adapter i plugged it into my ev3 robot and its still write there "Wifi not available", what should i do to make it works? i followed this video: https://www.youtube.com/watch?v=Op3lDRRN-Ck
and its seems like its works for him, please respond fast as possible.

dlech, 2021-07-02:
Please see https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles for a list of supported and known working Wi-Fi dongles.

ZonEmOToRY, 2021-07-02:
can i add a adapter that isnt in the list? there is any instruction to do that?

dlech, 2021-07-02:
The device in the video is already on the list. You probably have the v2 which is already described in the *Edimax Alert* section of that page.

ZonEmOToRY, 2021-07-02:
No, i dont have same adapter like its, i have d-link n300 Nano usb adapter

ZonEmOToRY, 2021-07-02:
I just followed about whats he doing

dlech, 2021-07-02:
Then feel free to list it under the not working adapters on that page. Be sure to include the USB vendor and product ID from `lsusb`.

ZonEmOToRY, 2021-07-02:
Ok how long its can take to be added to the working list?

ZonEmOToRY, 2021-07-02:
D-Link Wireless N300 Nano USB Adapter

dlech, 2021-07-02:
First, you need to know what the VID and PID is for the device, then you have to see if there are any existing Linux drivers for it, then you have to compile the driver against the ev3dev kernel. There are some other issues here where people have done that.

ZonEmOToRY, 2021-07-02:
USB\VID_0BDA&PID_818B&REV_0200
USB\VID_0BDA&PID_818B

this is what i found, https://www.realtek.com/en/component/zoo/category/rtl8192eu-software

ZonEmOToRY, 2021-07-02:
Its not for linux only windows so i cant continue

ZonEmOToRY, 2021-07-02:
how much time its will take to add it?