ev3dev/ev3dev #1503: ASUS N10 nano not working?

zeskeertwee, 2021-07-13:
I checked the [supported USB wifi dongle list](https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles), and the N10 nano was listed there, but when i plug it in the wifi menu says 'Not availible'...

The dongle does work with my desktop, which also runs linux
```lsusb``` shows it as
```0b05:18f0 ASUSTek Computer, Inc. Realtek 8188EUS [USB-N10 Nano]```

**Desktop (please complete the following information):**
Pop!_OS 21.04 x86_64
Ethernet desktop / wifi ev3dev
Fish terminal

**Robot (please complete the following information):**
EV3
Kernel: 4.14.117-ev3dev-2.3.5-ev3

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165343359A
BOARD_INFO_TYPE=main
```

dlech, 2021-07-16:
Manufacturers tend to put the same name on different devices. `0b05:18f0` is the number that really matters. So you can search for a Linux driver for `0b05:18f0` and try to compile it yourself or buy a different dongle.

Noc1a, 2025-02-27:
Probably a bit late, but I managed to solve it with https://github.com/ev3dev/ev3dev/issues/1442

zeskeertwee, 2025-03-11:
That's great! Thank you!

closed 2025-03-11 by zeskeertwee