ev3dev/ev3dev #1504: EV3 will not connect to the internet

nab138, 2021-08-03:
**What are you trying to do?**

I am trying to Connect my ev3 to the internet

**What did you expect to happen**

The program should have said Online and connect to the internet after I hit connect

**What actually happened**

The program said connected instead of online and did not connect to the internet. I could ssh but running `ping google.com` resulted in `Name or service not known`

**Desktop (please complete the following information):**
 - OS and version: Linux Mint 20.2 Uma Cinnamon 
 - Connection Type: Bluetooth (I Tried USB but it didnt work at all, I couldn't even ssh)
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165360E7F8
BOARD_INFO_TYPE=main
```

imposiblaa, 2021-08-24:
I'm having the same problem on linux, I know I got it working in April but don't know how I did it.

nab138, 2022-05-15:
This is still an issue- I have found that I can use `ping 1.1.1.1` with success so it seems to an issue with DNS resolution.

nab138, 2022-05-21:
For anyone having the same issue-
I edited `/etc/resolv.conf` with `sudo nano /etc/resolv.conf`
Commented all lines in there by adding `# ` to the start of them
Added two lines: `nameserver 8.8.8.8` and `nameserver 4.4.4.4`
`ping google.com` now works!

closed 2022-05-21 by nab138