ev3dev/ev3dev #1505: Initialising Sensor throws "OSError: [Errno 19] ENODEV: "

JohnGottiGe, 2021-08-19:
**What are you trying to do?**

I'm trying to use any sensor at the EV3 with Micropython. But I can't initialise the sensors, doesn't matter which sensor I try to use.

After the standard import statment

from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color

I'm trying to initialise a touch sensor for example, but it does not work.

Error is thrown at the following line:

_touch = TouchSensor(Port.S1)

What I have done so far?
Try to run it with two different Laptops --> did not work.
Reinstall the VS Code --> did not work
Reflash SD Card  --> did not work
Checked the Sensor in Device Browser --> here I can see the sensor and can check the actual value.

Install the DEBIAN Strech EV3DEV image is working

dlech, 2021-08-19:
Is the touch sensor an EV3 sensor or an NXT sensor?

JohnGottiGe, 2021-08-19:
It is an NXT sensor. I'm on vacation and I have only this one with me :-)
But the documentation says it should work also with this kind of sensor.

dlech, 2021-08-19:
There is a different module for NXT sensors.

```python
from pybricks.nxtdevices import TouchSensor
```

JohnGottiGe, 2021-08-19:
Oh, this information I did not see. I will try it asap.
Thanks a lot.

JohnGottiGe, 2021-08-19:
Hi David,
thanks a lot. Such a simple misstake I have done. With the right modul it is working perfect.
Thanks a lot

closed 2021-08-19 by JohnGottiGe