ev3dev/ev3dev #1509: binfmt_misc - cannot create register

johncblacker, 2021-09-01:
**Question**

How do you ...?
create the register file in /proc/sys/fs/binfmt_misc?

**Background Information**

I'm trying to add the register file to the binfmt_misc directory per closed issue #893, but it won't let me do it.  I keep getting "No such file or directory" error
I am trying to ... because I want to build a robot that ...
I want to run the hello output file

**Screenshots**
If applicable, add screenshots to help explain your question.
compiler@de187ca05123:/proc/sys/fs/binfmt_misc$ echo ':arm:M::\x7fELF\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x28\x00:\xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff:/usr/bin/qemu-arm-static:' > /proc/sys/fs/binfmt_misc/register
bash: /proc/sys/fs/binfmt_misc/register: No such file or directory
**Desktop (please complete the following information):**
 - OS and version: [e.g. Window 10 1809, macOS 10.14]
 - Windows 10 pro, build 19043
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - n/a
 - Software: [e.g. Visual Studio Code, PuTTY]
 ev3cc docker
**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
 - armhf
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

johncblacker, 2021-09-02:
So, what led me to raise this issue was that I was following your tutorial/recipe for building an armhf program with the cross-compiler and I tried to run it in the docker container as in the tutorial, but it doesn't run.  So, the comment there is that one has to go through the "register" in binfmt_misc steps in order to run the armhf executable in the container.  I googled that and found a number of hits, but whenever I tried them they didn't work - I wasn't able to create the "register" file(?).  Since you folks are the creator of this tutorial and repository I figured you'd be able to tell me how to do it.  The longer-term goal is to be able to create and armhf package for google's mediapipe from the current source, but for that I need bazel to build it and that means I have to build the latest bazel from sources, as well.  A lofty goal I'm sure and maybe not doable; but, others claim to have done it and I can attest to that having successfully used a version (old one) of mediapipe on my rPi 3b+.  I've also gotten a bazel version (older one) for armhf and on my rPi was successful at building mediapipe; but at runtime it ends up with an import error when I try using it in a python program.  Trying to change different build parameters to make it work isn't a good approach on the rPi because a mediapipe build takes 4+ hours! That's a long time to wait to find out that your "fix" didn't work.  I'm still trying to wrap my head around the idea of cross-compiling with bazel in a container...I'm not sure if I need and armhf version of bazel or one that's built for the x_64 architecture (I'm on win 10 pro running docker).  If I need an armhf version of bazel, then I have to build it and it has to be able to execute in the container.  Now, maybe my container should be a raspbian os, but an armhf version isn't going to run in a docker container on win10 either; so that leaves me with the x-86/64 raspbian (jessie, I think) running in a container with a cross compiler for the c++ code (java and python code shouldn't be an issue).  Well, that's the long story...I'd appreciate any help or direction you could provide.

dlech, 2021-09-02:
What OS is docker running on?

dlech, 2021-09-03:
Are you using Windows Subsystem for Linux?

johncblacker, 2021-09-03:
I'm not trying to run my commands in WSL,

dlech, 2021-09-03:
Docker for Windows should not require any manual steps to get Arm code to run inside the docker containers. The binftm_misc suggestion was for the older Docker Toolbox that ran in a VirtualBox VM (essentially running on Linux rather than Windows).

dlech, 2021-09-03:
What is the actual error you are getting?

johncblacker, 2021-09-03:
Look above under "screenshots" there's the output I get when issuing the echo command.

johncblacker, 2021-09-03:
> Docker for Windows should not require any manual steps to get Arm code to run inside the docker containers. The binftm_misc suggestion was for the older Docker Toolbox that ran in a VirtualBox VM (essentially running on Linux rather than Windows).

Be that as it may, when I try to run the hello world app, it says it can't execute because of executable format error or some such, I'd have to try it again to get the exact message output when I try to run the hello world executable.