ev3dev/ev3dev #1511: Transfert file from PC to EV3

gaminoi, 2021-09-14:
I'm trying to transfer Edimax EW-7811Un V2 WiFi dongle driver from my PC Linux to EV3 :
http://msdl.cs.mcgill.ca/people/rparedis/sources/wifi-dongle-EV3

EV3 is connected with USB, I can open an ssh terminal with vscode.
I manage to ping the EV3 from PC at 10.42.0.3, my PC is on 192.168.1.72, not on the same network ??
I use scp command from PC but "connection refused"

**What actually happened**
ssh: connect to host 10.42.0.3 port 22: Connection refused
lost connection

**Source Code**
gaminoi@PC:/home$ scp 8188eu.ko robot@10.42.0.3:/lib/modules/4.14.117-ev3dev-2.3.5-ev3/drivers/net/wireless

**Robot**
 - Device: EV3
 - Versions: 
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88

**Background Information**
I want to make my Edimax EW-7811Un V2 WiFi dongle work on the EV3

dlech, 2021-09-14:
Try replacing `10.42.0.3` with `ev3dev.local` or the IP address that appears on the screen of the EV3.

gaminoi, 2021-09-15:
`10.42.0.3` is the IP address that appears on the EV3, I can ping it from pc.
If I replace `10.42.0.3` with `ev3dev.local` :
```
ssh: Could not resolve hostname ev3dev.local: Name or service not known
lost connection
```

dlech, 2021-09-15:
Can you connect with `ssh robot@10.42.0.3`?

Sometimes SSH doesn't work because of a corrupted or defective SD card.

gaminoi, 2021-09-15:
I managed to transfert the file using the command :
`gaminoi@PC:/home$ scp 8188eu.ko robot@10.42.0.3:/lib/modules/4.14.117-ev3dev-2.3.5-ev3/drivers/net/wireless`
I needed to install `dnsmasq` package and to modify permissions of the file replaced on EV3.

Yes the connection with `ssh robot@10.42.0.3` works today with `dnsmasq` package. It didn't yesterday.
I just had to enter `sudo modprobe 8188eu` on EV3 and my dongle works !!

Thank you dlech for your help !

closed 2021-09-15 by gaminoi