ev3dev/ev3dev #1517: How do you create a sensor device for MicroInfinity CruizCore XG1300L in micropython?

LeeJenkins, 2021-10-30:
**Question**

How do you create and use a sensor device for MicroInfinity CruizCore XG1300L in micropython?

**Background Information**
I have a XG1300L plugged into port 3. I have a simple python3 program that successfully uses the sensor and prints angle values.

Now I want to use the sensor in a micropython program, but I cannot get it working.

**Screenshots**

The python3 program (working):
```
#!/usr/bin/python3

from ev3dev2.sensor import *
import time

# in the ev3dev brickman interface, navigate to
#     Device Browser > Sensors
# there you will find an entry for the mi-xg1300l
# create the Sensor with the string reported there.
uis = Sensor("i2c-legoev35:i2c1")

print("uis.address",uis.address)
print("uis.bin_data_format",uis.bin_data_format)
print("uis.decimals",uis.decimals)
print("uis.driver_name",uis.driver_name)
print("uis.modes",uis.modes)
print("uis.mode",uis.mode)
print("uis.units",uis.units)

for i in range(33):
    print( uis.value(0) / 100 )
    time.sleep(1)
```

Micropython program using I2CDevice. https://docs.pybricks.com/en/stable/iodevices/i2cdevice.html
The program generates "OSError: [Errno 5] EIO" on the call to device.read():
```
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.iodevices import I2CDevice
from pybricks.parameters import Port

# Initialize the EV3
ev3 = EV3Brick()

# Initialize I2C Sensor
CruizCore_XG1300L_I2C_address    = 0x02
CruizCore_XG1300L_read_angle_register = 0x42
GyroPort = Port.S3
print( "create an I2C device on port",GyroPort)
print( "    at I2C address %x"%CruizCore_XG1300L_I2C_address )
device = I2CDevice( GyroPort, CruizCore_XG1300L_I2C_address >> 1 )

print( "read one byte from register %x"%CruizCore_XG1300L_read_angle_register )
print( device.read( CruizCore_XG1300L_read_angle_register, 1 ) )
```

Micropython program using Ev3devSensor. https://docs.pybricks.com/en/stable/iodevices/ev3devsensor.html
The program generates "OSError: [Errno 5] EIO" on the call to Ev3devSensor():
```
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.parameters import Port
from pybricks.iodevices import Ev3devSensor

# Initialize the EV3
ev3 = EV3Brick()

# Initialize Sensor
port = Port.S3
print( "create Ev3devSensor on",port )
sensor = Ev3devSensor( port )

print( "read sensor angle" )
angle = sensor.read('ANGLE')
print(angle)
```

**Desktop (please complete the following information):**
 - OS and version: Window 10
 - Connection Type: WiFi connection to EV3
 - Software: edit/run files over ssh connection

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: 
```
$ uname -a
Linux tiger 4.14.117-ev3dev-2.3.5-ev3 #1 PREEMPT Sat Mar 7 12:54:39 CST 2020 armv5tejl GNU/Linux
```

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
$ ev3dev-sysinfo -m
<!-- Copy everything between these lines -->
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535492B5
BOARD_INFO_TYPE=main
```
<!-- Copy everything between these lines -->
```

dlech, 2021-10-30:
Since this is an issue with Pybricks, it would be better to report it at https://github.com/pybricks/support/issues.

The `Ev3devDevice` case looks like a bug in Pybricks. For some reason, 3rd party I2C sensors are excluded from `Ev3devDevice` ([source](https://github.com/pybricks/pybricks-micropython/blob/1473c4ba649fd38d07731c6104f66bf43b083d38/lib/ev3dev/src/ev3dev_stretch/lego_sensor.c#L84-L129)). Otherwise that would be the ideal way to use it.

For the `I2CDevice` case, you probably need to go through the Brickman menu and manually change input port 3 to `other-i2c` before running the program.

LeeJenkins, 2021-10-30:
Thanks, David. I will post it there.