ev3dev/ev3dev #1519: Bluetooth connection errors

TheAirBlow, 2021-11-16:
**What are you trying to do?**
Connect a EV3 with `MicroPython` (based on ev3dev) firmware (behavior same on latest original `ev3dev` firmware!) to my Linux laptop by using Bluetooth

**What did you expect to happen**
Simply connect

**What actually happened**
It disconnects shortly after, reconnection on Linux laptop results in the same thing.

Connection on EV3 connects and then shows a GDBUS Error: Operation Unavailable

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: Arch Linux (5.15.2-arch1-1)
 - Connection Type: Bluetooth
 - Software: **doesn't matter**

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: **latest MicroPython firmware, version unavailable as it's impossible to connect**

dlech, 2021-11-16:
Can you please give the exact steps to reproduce the problem? FYI, the *Connect* button on the Bluetooth screen on the EV3 is only for connecting keyboards and game controllers. It doesn't work for anything else, like connecting to a computer.

If you are using Pybricks, please see https://pybricks.com/ev3-micropython/messaging.html for instructions.

dlech, 2021-11-16:
Yes, this is expected. The EV3 doesn't provide any generic services like a keyboard or Bluetooth speaker, so connecting like this doesn't do anything useful. If you want a network connection, try https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/.

geoffpaulsen, 2021-12-04:
I too can't "see" the ev3dev device when trying to pair via bluetooth.
Enabling tethering on chrombook or on EV3dev didn't seem to help.

EV3 is running Bickman 0.10.3,
Chrombook is a amd64 chromebook.