ev3dev/ev3dev #1520: How do I make the ev3 detect my plugged-in HiTechnic NXT Infrared Seeker (old version, not V2) (part # MS1042)?

MichaelBonnet, 2021-12-12:
**What are you trying to do?**

I am trying to somehow connect the old version of the [HiTechnic NXT Infrared Seeker, part number MS1042](https://brickipedia.fandom.com/wiki/MS1042_Infrared_Seeker). I have been able to find vanishingly little information on this module.

**What did you expect to happen**

I expected the ev3 to at least notice that there is a device connected to the port.

**What actually happened**

The brick does not detect anything plugged into the port I have plugged the IR seeker into.

On the ev3 brick itself, Device Browser -> Sensors reveals just one sensor connected (a LEGO Ultrasonic Sensor), when the IR seeker is fully and securely plugged in. 

Error when running the code below:

```
Starting: brickrun --directory="/home/robot/Project_3" "/home/robot/Project_3/main.py"
----------
Traceback (most recent call last):
  File "/home/robot/Project_3/main.py", line 21, in <module>
    ir = Sensor(INPUT_2, driver_name='ht-nxt-ir-seek-v2')
  File "/usr/lib/python3/dist-packages/ev3dev2/sensor/__init__.py", line 78, in __init__
    super(Sensor, self).__init__(self.SYSTEM_CLASS_NAME, name_pattern, name_exact, **kwargs)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 223, in __init__
    chain_exception(DeviceNotFound("%s is not connected." % self), None)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 54, in chain_exception
    raise exception from cause
ev3dev2.DeviceNotFound: Sensor(ev3-ports:in2) is not connected.
----------
Exited with error code 1.
```

**Source Code**

Here is the program I am trying to run.

```
#!/usr/bin/env python3

###############
### IMPORTS ###
###############

from ev3dev2.sensor import Sensor
from ev3dev2.motor import LargeMotor, OUTPUT_A, OUTPUT_B, SpeedPercent, MoveTank, MoveSteering
from ev3dev2.sensor import INPUT_1, INPUT_2, INPUT_3, INPUT_4, Sensor
from ev3dev2.sensor.lego import UltrasonicSensor
from ev3dev2.led import Leds

#############################
### SENSOR/ACTUATOR SETUP ###
#############################

steer_pair = MoveSteering(OUTPUT_A, OUTPUT_B)
tank_pair = MoveTank(OUTPUT_A, OUTPUT_B)

# Ultrasonic sensor setup
us = UltrasonicSensor(INPUT_1)
us.MODE_US_DIST_CM = 'US_DIST_CM'

# Infrared seeker setup
ir = Sensor(INPUT_2, driver_name='ht-nxt-ir-seek-v2')
ir.mode = 'DC'
```

Variations of 'ht-nxt-ir-seek-v2' such as v1, removing the v2, seek -> seeker, do nothing for me. Fundamentally, it is failing to recognize the device as connected. I'm not sure where to go from here.

**Screenshots**

N/A

**Desktop:**
 - OS and version: macOS X Big Sur 11.6
 - Connection Type: USB
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165355464C
BOARD_INFO_TYPE=main
```

**Background Information**

I am trying to use the IR seeker to detect the direction of an infrared emitter relative to the heading of a robot that also uses an ultrasonic sensor to detect the distance to objects. Together, these sensors give me the direction and distance information I need to make my ev3 act like a [Braitenburg vehicle](https://en.wikipedia.org/wiki/Braitenberg_vehicle).

MichaelBonnet, 2021-12-12:
To clarify - in the Device Browser on the brick itself, Sensors shows just the ultrasonic sensor plugged into port 1 - `lego-ev3-us at ev3-ports:in1`.

Under Ports > ev3-ports:in2 actually shows  status of `nxt-i2c`, which tells me it is seeing that something is plugged in, but is not recognizing it as a sensor.

MichaelBonnet, 2021-12-13:
More information gained: In previous projects using this module, the device name (in a CPP program) was `NXT_IR_SEEKER`. Perhaps there's a way to get a driver for it?

MichaelBonnet, 2021-12-13:
Writing a driver probably isn't in my wheelhouse, both skill-wise and time constraint-wise, so I've begun attempting to use the raw i2c:

```
# Infrared Seeker setup
from smbus import SMBus

# Open i2c bus 4 (plugged in on Port 2) and read one byte from address 0x01, offset 0
bus = SMBus(4)
while True:
    print("Reading ...", file=stderr)
    for i in range(0, 10):
        b = bus.read_byte_data(0x01, i)
        print(b, file=stderr)
    print("End of read", file=stderr)

bus.close()
```
The issue is that I get the same readings, no matter what area my emitter is in (using a TV remote with a button held down and a 940nm infrared LED):

```
Reading ...
253
86
49
46
50
32
32
32
72
105
End of read
Reading ...
253
86
49
46
50
32
32
32
72
105
End of read
```
This should, in theory, behave like the NXT Infrared Seeker v2, but I'm not sure how to extract that behavior/information, or if I'm even pulling the right information/translating the extracted information into the format I would expect.

dlech, 2021-12-13:
Do you have documentation on the I2C registers of the sensor? Usually the values of interest start at register 0x42.

And I think you can use `read_i2c_block_data()` to read multiple bytes at a time, IIRC. Tip: use `struct.unpack()` to decode the bytes read.

MichaelBonnet, 2021-12-13:
I've only been able to find documentation [on the V2 version](https://www.robotshop.com/media/files/pdf/introduction-nsk1042.pdf). I had been hoping that the usage would be similar, but to no avail after thumbing through quite a few addresses:

```
# Infrared Seeker setup
from smbus import SMBus
import struct

# Open i2c bus 4 (seeker is plugged into port 2) and read bytes
bus = SMBus(4)
# while True:
print("Reading ...", file=stderr)
for i in range(0, 0xFF):
    try:
        # b = bus.read_byte_data( (0x41 + i), 0)
        b = bus.read_i2c_block_data( (0x00 + i), 0)
        # b = struct.pack('<l', b)
        # b_prime = struct.unpack('<l', b)
        # b_prime = struct.unpack('<l', ''.join([chr(i) for i in b[:4]]))[0]
        print("Address "+str(hex(0x00+i))+" is good", file=stderr)
        print(b, file=stderr)
        # print(b_prime, file=stderr)
    except IOError:
        print("Address "+str(hex(0x00+i))+" not good", file=stderr)
print("End of read", file=stderr)

bus.close()
```

As you can see, I tried to use the `struct.unpack()` method you mentioned, but I couldn't quite figure it out.

```
Starting: brickrun --directory="/home/robot/Project_3" "/home/robot/Project_3/main.py"
----------
Reading ...
Address 0x0 not good
Address 0x1 is good
[253, 86, 49, 46, 50, 32, 32, 32, 72, 105, 84, 101, 99, 104, 110, 99, 73, 82, 32, 68, 105, 114, 46, 32, 0, 0, 0, 0, 0, 0, 0, 0]
Address 0x2 not good
Address 0x3 not good
Address 0x4 not good
Address 0x5 not good
Address 0x6 not good
Address 0x7 not good
Address 0x8 not good
Address 0x9 not good
Address 0xa not good
Address 0xb not good
Address 0xc not good
Address 0xd not good
Address 0xe not good
Address 0xf not good
Address 0x10 not good
Address 0x11 not good
Address 0x12 not good
Address 0x13 not good
Address 0x14 not good
Address 0x15 not good
Address 0x16 not good
Address 0x17 not good
Address 0x18 not good
Address 0x19 not good
Address 0x1a not good
Address 0x1b not good
Address 0x1c not good
Address 0x1d not good
Address 0x1e not good
Address 0x1f not good
Address 0x20 not good
Address 0x21 not good
Address 0x22 not good
[... 201 lines cut ...]
Address 0xec not good
Address 0xed not good
Address 0xee not good
Address 0xef not good
Address 0xf0 not good
Address 0xf1 not good
Address 0xf2 not good
Address 0xf3 not good
Address 0xf4 not good
Address 0xf5 not good
Address 0xf6 not good
Address 0xf7 not good
Address 0xf8 not good
Address 0xf9 not good
Address 0xfa not good
Address 0xfb not good
Address 0xfc not good
Address 0xfd not good
Address 0xfe not good
End of read
----------
Completed successfully.

```
Only one register being readable at all seems ... fishy.

dlech, 2021-12-13:
It looks like `read_i2c_block_data(address, register, length)` is missing the first parameter, 0x01, the I2C address of the sensor. And the length is 0, so it won't read any data.

MichaelBonnet, 2021-12-13:
We managed to figure it out. We finally managed to find a project that told us what drivers to use - specifically, these two files([1](https://github.com/botbench/rdpartyrobotcdr/blob/master/drivers/hitechnic-irseeker-v1.h), [2](https://github.com/botbench/rdpartyrobotcdr/blob/master/hitechnic-irseeker-v1-test1.c)) told us what we needed to know.

Relevant, working code:

```
from time import time, sleep
from smbus import SMBus
from sys import stderr

# Infrared Seeker setup
# Open i2c bus 4 (use port # + 2 -- seeker is plugged into port 2)
bus = SMBus(4)

# Takes the direction reading (0-9) from the IR seeker
# via direct, raw i2c interactions
# and returns it.
# Takes the direction reading (0-9) from the IR seeker
# via direct, raw i2c interactions
# and returns it.
def get_ir_direction(bus):
    direction = bus.read_byte_data( 0x01, 0x42 )
    return direction

while True:
    # Sensor Reading Testing
    print( get_ir_direction(bus), file=stderr ) # prints to stderr so that we can see the readings on the computer screen rather than ev3 screen.
```

Thank you @dlech for pointing us in the right direction!

closed 2021-12-14 by MichaelBonnet