ev3dev/ev3dev #1524: Unable to find the driver code for the HiTechnic IrSeeker V2 Sensor

Eisverygoodletter, 2022-01-22:
**Question**

I would like to find the code for the HiTechnic IrSeeker V2 Sensor's driver.

**Background Information**

After reading [some info on the HiTechnic IrSeeker V2 Sensor](https://www.robotshop.com/media/files/pdf/introduction-nsk1042.pdf), I noticed that it said that channels 2, 4, 6, 8 were somehow "interpolated". I would like to know how they are "interpolated" by the sensor's driver, or overall how it is processed. However, I can't find the files for the driver in https://github.com/ev3dev/ev3-kernel or https://github.com/ev3dev/ev3-kernel/tree/ev3dev-buster/drivers (using github's "go to file" search button). Am I looking in the wrong place? And is there some kind of naming convention for driver code?

I am trying to read about how the driver works because I want to build a robot that finds its approximate distance from an IR ball using the Ir Seeker sensor, but I do not understand how the driver calculates the values.

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev-stretch
 
small note: I'm a beginner at messing with drivers and may actually have no idea how drivers are written.

dlech, 2022-01-22:
The code for this sensor is here: https://github.com/ev3dev/lego-linux-drivers/blob/2c05d904e82ee62a5425ba244256a9cf7928f5a2/sensors/nxt_i2c_sensor_defs.c#L746-L847

Eisverygoodletter, 2022-01-22:
Thanks! seems like the "interpolating" is done in the sensor is not the driver, but I found a way around my problem by checking the `AC` direction and `AC-ALL` for the signal strength measurement I wanted. There's a problem with this though and I think I'll create a new issue for it.

closed 2022-01-22 by Eisverygoodletter