ev3dev/ev3dev #1525: HiTechnic IrSeeker V2 `AC-ALL` mode is documented to return `0-9` for signal strength, but actual amount goes above `80`.

Eisverygoodletter, 2022-01-22:
note: continuing from #1524 
**Describe the bug**
Documentation says `AC-ALL` signal strengths should return `0-9`. In reality, returns `0 - ~100`

**To Reproduce**
Steps to reproduce the behavior:
1. Set the mode to `AC-ALL`
2. check `value1` `value2` `value3` ... to `value5` while placing an IR ball in front of the sensor. (for example, value 3 corresponds to the sensor's "forward" direction, `value1` is the left direction, etc.)
3. see that the values go over the range `0-9`, which is what the documentation says the values should be in range of.
```
Please paste a simple program to reproduce the problem here!
// note: this is a class I wrote that works with [the ev3dev cpp library](https://github.com/ddemidov/ev3dev-lang-cpp)

// in a .hpp file
class irseeker_sensor : public i2c_sensor{
    public:
        irseeker_sensor(address_type address = INPUT_AUTO);
        static char mode_irseeker_ac_all[]; // defined in another .cpp file as AC-ALL
        std::vector<int> ac_all(bool do_set_mode = true){
            if(do_set_mode) set_mode(mode_irseeker_ac_all);
            return {
                value(0),
                value(1),
                value(2),
                value(3),
                value(4),
                value(5)
            };
        }
};

// inside a main.cpp
// note - this is to test `value1`. In this example, I would place the ball on the left of the robot so that the `value1` direction would detect the ball
int main() {
    irseeker_sensor myIrSeeker = irseeker_sensor();
    std::vector<int> acValues = myIrSeeker.ac_all();
    std::cout << acValues[1] << '\n';
    std::this_thread::sleep_for(std::chrono::seconds(10)); // some time for you to read the output before program ends
}
```

**Expected behavior**
the program prints out a number from 0-9
**Problem**
the program prints out numbers going from 0 - ~100 (unsure what the upper limit is)

**Robot (please complete the following information):**
 - Device: ev3
 - Versions: ev3dev-stretch

**Additional context**
1. This problem doesn't cause any bugs or crash the program. The documentation just needs to be updated so that it doesn't say it would return a range from `0-9`

2. the `value0` from `AC-ALL` corresponds to the direction the ball should be in. This part of the `AC-ALL` documentation is correct, as `value0` will always return some number in `0-9`

3. a signal strength of 80 is approximately when the ball is 15 centimetres from the sensor.
**My Suggestion:** 
remove text from documentation such as https://github.com/ev3dev/lego-linux-drivers/blob/2c05d904e82ee62a5425ba244256a9cf7928f5a2/sensors/nxt_i2c_sensor_defs.c#L746-L847 saying that `AC-ALL` values 1-5 would return signal strengths of `0-9`

dlech, 2022-01-22:
Yes, you can make a pull request if you want to.

Eisverygoodletter, 2022-01-23:
made a pull request [here](https://github.com/ev3dev/lego-linux-drivers/pull/84)

closed 2022-03-05 by Eisverygoodletter