ev3dev/ev3dev #1527: Connect to EV3 Brick USB and Bluetooth issues

New2CS, 2022-02-15:
- ev3dev version 1.2.0 used in Visual Source Code
- I'm trying to consistently connect my EV3 brick to my computer.  The brick runs my codes once it's connected, but I can't connect it multiple times.  I plug the USB (in several ports) in and the laptop recognizes a new device. But VSC won't connect.  I get a timeout error
-  failed to connect: connect EHOSTUNREACH
- I get the same error when I try to connect using Bluetooth

-  I do not get this error when using the original Mindstorm firmware.
- 
- all user guide instructions I've found don't resolve this issue. The only way I can connect is to 
- right-click connect to a different device
- Click I don't see my device...
- create a new device name with the same IP address as previously created devices
- Then the brick connects, but as soon as I run code or disconnect from the computer I cannot reconnect.  I currently have a list of 8 different names that are all the same device,

dlech, 2022-02-15:
What operating system do you use? Can you give the exact steps to reproduce the problem?

New2CS, 2022-02-16:
im on a Windows 10 device

Open up VScode, uninstall EV3DEV and uninstall Mindstorm Micropython extensions
close VScode and reopen
Reinstall EV3Dev and Mindstorm extensions

Turn on EV3 Brick and wait for it to boot
open up a file
plug USB into Brick and into computer
computer makes a noise indicating a USB has be inserted
open Device Browser and right click on device 
click reconnect 
light turns yellow
after a period of time the Error I stated before shows up
the same issue occurs when I try to connect with Bluetooth

to get around this I can create a new device with the IP address, but this only works for one connection at a time.