ev3dev/ev3dev #1533: EV3 - pthread_setschedparam doesn't work launching the program from brickman

cho934, 2022-05-07:
Hi,

I'm trying to change the priority of the thread in cpp.

When I'm connected with ssh with the user "robot", everything is working great, SCHED_FIFO is set and each thread has his own priority.
Now I need to launch the same program from Brickman, and this command "pthread_setschedparam" return an error.

To be able to set SCHED_FIFO, I modified /etc/security/limits.conf and set for the user robot:
```
robot hard rtprio 99
robot soft rtprio 99
```

What is wrong ?
Brickman doesn't use the user "robot" ?

**Source Code**
see here 
http://www.yonch.com/tech/82-linux-thread-priority
and here
https://stackoverflow.com/questions/10704983/operation-not-permitted-while-setting-new-priority-for-thread

**Robot (please complete the following information):**
 - Device: EV3

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           226.09
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534560D1
BOARD_INFO_TYPE=main
```

dlech, 2022-05-07:
When you run a program through Brickman or from the command line with `brickrun`, the program will already be running with a `nice` value of `-10` which seems good enough for most tasks. You could edit `/etc/systemd/system/console-runner@.service` (and reboot) to make it even higher (-20 max). This might be easier than trying to solve the thread priority issue with the same results.

> Brickman doesn't use the user "robot" ?

Yes, it does, using `su` and through `brickrun` so there may be some things about the environment that aren't quire the same. I'm not sure what could be making the difference in this case.

cho934, 2022-05-09:
Hi,

Thanks a lot for the explanation.
I solved the issue by editing /etc/systemd/system/console-runner@.service and adding:
```
LimitRTPRIO=90
```
and by increasing the parameter Nice to -20
Then no more error message using pthread functions like pthread_setschedparam()

closed 2022-05-09 by cho934