ev3dev/ev3dev #1534: ev3dev libraries on Docker image

r0bis, 2022-05-29:
**How to add ev3dev libraries/headers to docker cross-compile setup?**

I have assembled ev3dev on BrickPi (with Raspberry Pi 1 - chip BCM2835). I do want to cross compile and I have been able to download docker image:  
```
`ev3dev/debian-stretch-cross   latest    8e3b0aa8f243   4 years ago     732MB`
`ev3cc                         latest    8e3b0aa8f243   4 years ago     732MB`  
```
I access the compiler via `docker run` (`docker run --rm -it -v /home/r0bis/src/robotics/helloArm/:/src -w /src ev3cc`)  
The compiler can compile hello world program that I can then put on RPI and run successfully. However when I try to compile examples that include ev3dev specific libraries it fails. In the tutorial I do not find what would be the best way to add the missing libraries. I thought that the libraries/headers would be included in the docker image; maybe they are and I just do not know something. Help would be much appreciated.

**Screenshots**
```
compiler@3547e87f257e:/src$ arm-linux-gnueabi-gcc port.c -o port
port.c:8:17: fatal error: ev3.h: No such file or directory
 #include "ev3.h"
                 ^
compilation terminated.

```
**Desktop:**
 - Manjaro Linux 64 bit
 - Connection Type: SSH over ethernet
 - Software: Text editor, ssh/sftp connection; would like to set up VSCode eventually

**Robot:**
 - Device: BrickPi 3 (Raspberry Pi 1 B+ (BCM2835)
 - Versions: 

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi-generic-2020-04-10
Kernel version:     4.14.114-ev3dev-2.3.5-rpi
Brickman:           0.10.3
BogoMIPS:           697.95
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=0010
BOARD_INFO_MODEL=Raspberry Pi Model B Plus Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000d374d92b
BOARD_INFO_TYPE=main
```

dlech, 2022-05-29:
ev3dev doesn't ship with an `ev3.h` header. Also the `ev3cc` docker image isn't one of the official ev3dev docker images.  Which library are you trying to use?

In general though, instead of running the docker image directly, I would create a docker container using that image and then run a shell in the container and install whatever you need.

r0bis, 2022-05-29:
Sorry - should have said `ev3cc` is just a tag/shortname of the official image that I am running. `ev3dev/debian-stretch-cross   latest    8e3b0aa8f243   4 years ago     732MB` I will try to understand better about containers from your answer - as to how to run a shell in the container. My thought was - if I am using the container as a place where I do compiling, then I should have an easy enough way to get all the ev3dev libraries relevant to my RPI (1 in my case) on to the image. At the moment I just can't quite figure out how.

r0bis, 2022-06-02:
OK, I sorted out docker part. Code now compiles without error messages on docker and runs on Pi. brickpi3 is uncommented in ` /boot/flash/config.txt` 

`lsmod` shows:
```
evdev                  11758  4
brickpi3               21908  0
ev3_uart_sensor        13927  1 brickpi3
nxt_i2c_sensor         39176  1 ev3_uart_sensor
i2c_bcm2835             6465  0
lego_bus                4302  3 brickpi3,ev3_uart_sensor,nxt_i2c_sensor
spidev                  7156  0
lego_port_class         4573  2 brickpi3,nxt_i2c_sensor
servo_motor_class       7256  1 nxt_i2c_sensor
lego_sensor_class      10650  2 ev3_uart_sensor,nxt_i2c_sensor
brickpi3_battery        2766  0
```
but I cannot get example scripts to work and I do not know how to test these with C in a simplest way.  Motors, sensors and ports do not seem to be connected.  Any pointer would be appreciated.

dlech, 2022-06-02:
> ev3_tacho_init();

This sounds like you are using https://github.com/in4lio/ev3dev-c, so you might get better help if you ask there. I've never used that library.

r0bis, 2022-06-03:
> This sounds like you are using https://github.com/in4lio/ev3dev-c, so you might get better help if you ask there. I've never used that library.

Sorry, I did not know there was another library for using C with ev3dev. I will ask there, but if you could let me know what the alternatives are there for C, it would be much appreciated.

r0bis, 2022-06-03:
I understand that device (sensor/motor/port) needs to be detected to be found in "All device classes in Linux are found at `/sys/class`". The problem I have is that there is nothing for BrickPi3. 

For example on EV3 I can do 
```
cat /sys/class/lego-sensor/sensor0/address  
ev3-ports:in2
```
On RaspberryPi Brickpi3 lego-sensor/ is empty
```
ls  -l /sys/class/lego-sensor/
total 0
```
And same goes for `/sys/class/lego-port/`

dlech, 2022-06-03:
> ```
> 
> ```
> 
> 
>     
>   
> 
> When I run `dmesg | grep brick` there is no output and I cannot see anything like that if I scroll either.

Did you enable brickpi3 in `/boot/flash/config.txt` and reboot the Pi?

r0bis, 2022-06-03:
Yes, I did that. That did not give any messages related to brickpi3 in dmesg whatsoever. Then I could not load overlay either with `dtoverlay brickpi3` or with indicating the directory like below. (`dtoverlay` expects `/boot/overlays` or `/flash/overlays`)
```
root@ev3dev:~# dtoverlay -v -d /boot/flash/overlays/ brickpi3
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: delete_node(/__symbols__)
DTOVERLAY[debug]: Wrote 1449 bytes to '/tmp/.dtoverlays/0_brickpi3.dtbo'
DTOVERLAY[debug]: Wrote 1449 bytes to '/sys/kernel/config/device-tree/overlays/0_brickpi3/dtbo'
* Failed to apply overlay '0_brickpi3' (kernel)
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post
```
Then I created `/boot/overlays` and copied `/boot/flash/overlays/brickpi3.dtbo` into it and also created `/boot/config.txt` - that gave error messages in dmesg upon reboot:
```
[   74.240777] IPv6: ADDRCONF(NETDEV_CHANGE): enxb827eb74d92b: link becomes ready
[  150.361082] OF: overlay: Failed to apply single node @/soc/spi@7e204000/brickpi3
[  150.361128] OF: overlay: apply failed '/soc/spi@7e204000'
[  150.361159] create_overlay: Failed to create overlay (err=-22)
[  167.807058] OF: overlay: Failed to apply single node @/soc/spi@7e204000/brickpi3
[  167.807105] OF: overlay: apply failed '/soc/spi@7e204000'
[  167.807133] create_overlay: Failed to create overlay (err=-22)
[  382.735708] OF: overlay: Failed to apply single node @/soc/spi@7e204000/brickpi3
[  382.735757] OF: overlay: apply failed '/soc/spi@7e204000'
[  382.735786] create_overlay: Failed to create overlay (err=-22)
[  389.266778] OF: overlay: Failed to apply single node @/soc/spi@7e204000/brickpi3
[  389.266825] OF: overlay: apply failed '/soc/spi@7e204000'
[  389.266852] create_overlay: Failed to create overlay (err=-22)
[  711.968415] OF: overlay: Failed to apply single node @/soc/spi@7e204000/brickpi3
[  711.968463] OF: overlay: apply failed '/soc/spi@7e204000'
[  711.968492] create_overlay: Failed to create overlay (err=-22)
[  810.882021] OF: overlay: Failed to apply single node @/soc/spi@7e204000/brickpi3
[  810.882069] OF: overlay: apply failed '/soc/spi@7e204000'
[  810.882164] create_overlay: Failed to create overlay (err=-22)
```

dlech, 2022-06-03:
Since you have changed things, I would re-flash the SD card and try again. If it still doesn't work, have a look at https://www.raspberrypi.com/documentation/computers/configuration.html#troubleshooting to get some logs before changing anything.

r0bis, 2022-06-03:
Thanks, this was just in time. I re-downloaded the image, reflashed and after uncommenting brickpi3 line `/boot/flash/config.txt` i am now getting:
```
   8.323629] brickpi3 spi0.1: Address: 1
[    8.326031] brickpi3 spi0.1: Mfg: Dexter Industries
[    8.373228] brickpi3 spi0.1: Board: BrickPi3
[    8.399260] brickpi3 spi0.1: HW: 3.0.0
[    8.430060] brickpi3 spi0.1: FW: 1.4.8
[    8.436554] brickpi3 spi0.1: ID: 1AA0A515515035524E4A2020FF080122
```
So all I need to work out is how to correctly pass port/motor parameters using the library. 

Thanks @dlech

closed 2022-06-03 by r0bis