ev3dev/ev3dev #1535: USB Networking doesn't work.

COMHack20, 2022-06-23:
I tried connecting the EV3 with USB to the internet. I followed the official guide. And when I have plugged in the EV3, my laptop suddenly force shuts down the Wi-Fi and I cannot put it back on while the EV3 is still connected. I tried resetting network adapter settings and removed the EV3 device, then plugged back in and Windows reinstalled the driver.

The problem could be though, that when I reinstalled ev3dev, then obviously it's a new device in Windows.  Maybe the old ev3dev device in Windows is interfering in some way? IDK though. Any ideas?

snowfxll, 2023-11-18:
bro still nothing? it's been over a year now and nobody answered lol

dlech, 2023-11-18:
The official guide works for most people, so to figure out your problem you would need to figure out what is different about your computer compared to everyone else.

snowfxll, 2023-11-22:
well the guide is for windows 10.. apart from that, i don't really know. the thing is, whenever i plug the usb in my pc automatically tries to connect to the ev3 instead

Growflavor, 2023-12-08:
> well the guide is for windows 10 (i'm on 11).. apart from that, i don't really know. the thing is, whenever i plug the usb in my pc automatically tries to connect to the ev3 instead

just in case the following might help you troubleshoot further:

a) for me with ev3dev-stretch-ev3-generic the 'official guide' for usb internet sharing access:

 https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/ 

works for me on both windows 10 & 11 pro machines both before & after making updates
(for example, after updating the sources.list: see #1594 ) to do ML projects such as 
https://leenahere.github.io/blog/ml-with-ev3/

b) for me, SSH & VisualStudioCode connections will work for code transfers even if the internet sharing connection is broken...

c) & for my machines the sign of a _working_ usb internet sharing appears to be the EV3 LCD display showing an IPv4 192.168.137.xxx IP rather than something else like an 169.xxx.xxx.xxx IP)

d) RE:"whenever i plug the usb in my pc automatically tries to connect to the ev3 instead"  When booting EV3dev_stretch that does not occur for me (where as it will occur if I boot regular LEGO 1.10E firmware and use usb to connect to either windows 10 or 11) so that might be a clue for your further troubleshooting...