ev3dev/ev3dev #1537: NameError: name 'Warning' isn't defined

Allen-1216, 2022-07-26:
**Question**

I follow this download kociemba into ev3
```
$ sudo apt-get update
$ sudo apt-get install build-essential libffi-dev
$ cd ~/
$ git clone https://github.com/dwalton76/kociemba.git
$ cd ~/kociemba/kociemba/ckociemba/
$ make
$ sudo make install
```

**Background Information**
I am trying to kociemba algorithms because I want to build a rubik's cube solver robot

**Screenshots**

The terminal can run kociemba with the command

[image]

but
[image]
[image]
[image]

Error message

```
Traceback (most recent call last):
  File "/home/robot/ev3micropython_kociemba_test/main.py", line 9, in <module>
  File "/home/robot/ev3micropython_kociemba_test/kociemba/__init__.py", line 4, in <module>
NameError: name 'Warning' isn't defined
```
kociemba/__init__.py
[image]

Desktop :
 - OS: Window 11
 - Connection Type: USB
 - software: Visual Studio Code

Robot:
 - Device: EV3
 - Versions: ev3-micropython-v2.0

```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534AA73E
BOARD_INFO_TYPE=main
```

dlech, 2022-07-26:
MicroPython is not Python, so unless `kociemba` was written specifically for MicroPython, it probably isn't going to work. You could try https://github.com/ev3dev/ev3dev-lang-python instead.

Allen-1216, 2022-07-27:
thank you for your help @dlech 
I solved the problem using `system()`

closed 2022-07-27 by Allen-1216