ev3dev/ev3dev #1538: Meaning of `cross` in `ev3dev/debian-stretch-cross` Docker image

Eisverygoodletter, 2022-08-12:
**Question**
What does the `cross` in `ev3dev/debian-stretch-cross` mean?

**Background Information**
If I go to the [guide](https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/) about cross compiling with docker, it tells me to use the `ev3dev/debian-stretch-cross` image. However, if I visit [`ev3dev/docker-library` repository](https://github.com/ev3dev/docker-library/tree/master/ev3dev-stretch), there is no mention of `cross` being a valid variant (the only 2 mentioned are `base` and `generic`).

dlech, 2022-08-12:
It means the image is meant to be used for **cross**-compiling (as opposed to being used for creating a disk image which is what the docker-library repo is for).

Eisverygoodletter, 2022-08-12:
As a related question: would it be possible to install a newer version of CMake in the image? It currently has CMake version 3.7.2.
I've tried to make a dockerfile
```Dockerfile
FROM ev3dev/debian-stretch-cross:latest
RUN sudo apt-get update -y
RUN sudo apt-get install cmake
```
But it still installs CMake 3.7.2

closed 2022-08-12 by Eisverygoodletter